serialport/node-serialport #1336: Event loop frozen in 6.0.0-beta.2

espitall, 2017-09-14:
- __SerialPort Version__: 6.0.0-beta.2
- __NodeJS Version__: 1.6.10 (electron)
- __Operating System__ and __Hardware Platform__: Win 7 64bits

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: yes

### Summary of Problem

I am opening and closing the same port (without sending data) every second for testing purpose and after a while, the electron main loop seems frozen.
I have tried to find the root cause of this a it seems to be frozen on the EIO_AfterRead function on line
`WaitForSingleObject(baton->hThread, INFINITE);`
I have checked and the ExitThread(0) is called in the Read Thread but it seems that the WaitForSingleObject is not notfied.

### Steps and Code to Reproduce the Issue

reconbot, 2017-09-24:
Are you registering any data events? Example code would help us reproduce. Thanks!

reconbot, 2017-10-18:
I think this is related to #1363 and fixed in #1364 will have a beta out in a bit.

reconbot, 2017-10-20:
v6.0.1 should have a fix for this

reconbot, 2017-10-26:
v6.0.4 does have a fix. Please let me know if this solves your problem.

closed 2018-02-05 by reconbot