serialport/node-serialport #1337: %1 is not a valid Win32 application

Usey95, 2017-09-18:
- __SerialPort Version__: 4.0.7(johnny-five default) & 6.0.0beta
- __NodeJS Version__: 8.4.0
- __Operating System__ and __Hardware Platform__:  osx darwin-x64, arduino

Hello,
I'm using johnny-five & electron v1.7.5 to build an cross platform app. The app runs well when debugging(electron ., I can successfully run electron-rebuild), but crashes after build.
In mac, the app crashes once I link arduino. In windows, it throws the error "%1 is not a valid Win32 application......serialport.node".
I followed some (instructions)[https://www.bountysource.com/issues/47111100-error-1-is-not-a-valid-win32-application], but none of them can solve the problem.

goodhoko, 2017-09-18:
@Usey95 I've head this problem when packing my app on 64bit plaform to 32 bit plaform with electron-packager. This is because the native plugin (eg serialport) is not recompiled for the target platform. What fixed the problem for me was adding an afterCopy parameter when calling the packager like this:
```
const packager = require('electron-packager')
const rebuild = require('electron-rebuild').rebuild
packager({
    name: 'GoOutPrinterBridge',
    dir: './source',
    platform: 'win32',
    arch: 'ia32',
    //we must do a rebuild because serialport module is native
    afterCopy: [(buildPath, electronVersion, platform, arch, callback) => {
        rebuild({ buildPath, electronVersion, arch })
          .then(() => callback())
          .catch((error) => callback(error));
        }
    ]
})
```

I hope this will help you.

Usey95, 2017-09-18:
@goodhoko Thanks for your advice, it seems afterCopy is a possible hook to fix this. But since cross platform app can only be built by electron builder, for now I can't find a way to use it. :(

Usey95, 2017-09-19:
I solved the problem by replace the johnny-five internal serialport with the latest version!!
Now I can rebuild it correctly.

closed 2017-09-19 by Usey95