serialport/node-serialport #1339: Default parser different output on different versions

Softwareschmiede, 2017-09-19:
- __SerialPort Version__: 4.0.7 & 5.x
- __NodeJS Version__: 6.11.3 LTS
- __Operating System__ and __Hardware Platform__: Debian ARMv7

- Have you checked the right __version__ of the __api docs?__: Yes, Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No, No
- Are you using __Electron__ and have you checked the _Electron Docs_?: No, No

### Summary of Problem
As i used version 4.0.7 i got the following buffer result:
`<Buffer 55 00 0a 07 01 eb a5 00 00 80 08 01 81 38 3f 00 03 ff ff ff ff 47 00 8d>`

Now i'm using version 5.x and now i get the following buffer result:
`<Buffer b1 31 4c fc>`

As you can see the length is different. And the buffer on version 4.0.7 always starts with 55 (that's what is correct)

### Steps and Code to Reproduce the Issue
```
port = new SerialPort('/dev/ttyS3', { baudrate: 57600 });

port.on('error', function(err) {
    console.log('Error: ', err.message);
});

port.on('data', function(buf) {
    console.log(buf);
});
```

reconbot, 2017-09-19:
Which parser are you using? We did change how reads work but you need a
parser to guarantee how much data you receive at a time.

Softwareschmiede, 2017-09-19:
From your docu of version 4 i'm using the "raw parser", because i'm didn't specify any parser.
So i thought i version 5 i'm using the same parser, didn't i?

reconbot, 2017-09-19:
The "raw parser" is no parsing at all. It returns data as it's received. In
5x we receive it faster so you'll sometimes get more frequent but smaller
data events.

---
Francis Gulotta
wizard@roborooter.com

Softwareschmiede, 2017-09-20:
Hmmm i don't think that this is the problem.
Then my console should show something like this:

```
<Buffer 55 00 0a 07>
<Buffer 01 eb a5 00>
<Buffer 00 80 08 01>
<Buffer 81 38 3f 00>
<Buffer 03 ff ff ff ff>
<Buffer 47 00 8d>
```

But every 10 minutes i only get one package of 4 bytes.

Softwareschmiede, 2017-09-21:
I wrote a custom parser based on the delimiter parser which only pushes all chucks which are comming in. But the same problem. I only got one buffer of 4 bytes every ten minutes.

Any suggestions?

Softwareschmiede, 2017-09-22:
The problem was that "baudrate" changed to "baudRate". So you have to use camelcase not lowercase.

closed 2017-09-22 by Softwareschmiede

reconbot, 2017-09-22:
Ahh good catch. That is in the [upgrade guide](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/UPGRADE_GUIDE.md) but so much changed between versions it didn't even occur to me.