serialport/node-serialport #1340: RE-Establish connection to port after closed.

eshk12, 2017-09-19:
SerialPort Version: V5.0.0
NodeJS Version: V6.3.0
Operating System: WIN 10 64B

Hello,
Accordding to my last question [`close` event never fire](https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/1334). I was unabled to detected if the `COM` is disconnected so I have created my own way to detect it.

I have created timestamp, and checked it with `interval` every 1 sec to see if is connected.
when it's detect the `COM` is unplugged I have try to re-establish the connection be re-instact port with `SerialPort` like you'll see inside the code below.
When its try to reconnect I've get `Error: Access denied.`
there is a way to refresh or clean the cache, because I think the server still hold the connection when isn't closed propely.

I've also tried `port.close()` and it's throw me: `Error: Port is not open`.

```
var comPort = '\\\\.\\COM7',
    lastDataTime,
    lastresult,
    count = 0,
    lastDataTime,
    comStatus,
    error;
var port = new SerialPort(comPort, function (err) {
    if (err) {
        comStatus = false;
        return console.log('Error: ', err.message);
    }
});
const parser = port.pipe(new Readline());
port.on('open', function () {
    console.log('~Port is open.');

    parser.on('data', function (data) {
        comStatus = true;
        lastDataTime = Date.now();
        if (++count == 10) {
            count = 0;
            lastresult = data;
        }
    });
});

setInterval(function () {
    if (Date.now() - lastDataTime > 1000 || !comStatus) {
        comStatus = false;
        port.close();
        port = new SerialPort(comPort, function (err) {
            if (err) {
                error = 'Error: ' + err.message;
                return console.log(error);
            }
        });
    }
}, 1000);

app.get('/', function (req, res) {
    res.send((comStatus) ? lastresult : 'Disconnected - ' + error);
    console.log(lastresult);
})
```

Thanks!

fluentart, 2017-09-22:
port.on('close', function(err) {
    console.log('should reconnect now');
   connectMyPort();
});

Best to move all your serialport code into connectMyPort(), then it can recursively call itself when the port is closed.

reconbot, 2017-09-22:
You can also listen for data events right away, no need to wait for open.

reconbot, 2017-10-26:
I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-10-26 by reconbot

anwarhamr, 2018-01-02:
This is still an issue.

reconbot, 2018-01-02:
@anwarhamr mind opening your own issue with details of your setup?

anwarhamr, 2018-01-02:
A new issue or here?

reconbot, 2018-01-02:
New issue