serialport/node-serialport #1344: Serial port Polling is limited to vsync speed

ivanseidel, 2017-09-24:
- __SerialPort Version__: 5.0.0
- __NodeJS Version__: 8.x
- __Operating System__ and __Hardware Platform__: OSx/Unix and Windows

### Summary of Problem

Writing to SerialPort is fast, and not a problem.

However, reading data from SerialPort is limited to the `vsync` of the monitor because uv_pool internally uses that to pool for changes.

That means that, a device can only send/receive serial 60 packets or 30 serial packets serially one after the other although communication can go up to 500Kb/s without any problem.

It seems to me, that it should be using some sort of socket piping when reading from the Serial, but that seems unclear as no one has done that before.

Looking trough libuv, only this thread stated some sort of problem like that, but is not really related: https://github.com/joyent/libuv/issues/1246

### Steps and Code to Reproduce the Issue
1) Make a loop back on any hardware, and log the speed taken to read from the SerialPort
2) Increase what's being written, and you will see a clear historgram in times taken to receive the data:

~16.66ms/~33.33ms/~50ms/...

That is cause because Reading is synchronized with the vsync from the monitor, and it shouldn't , as SerialPort is unrelated with the screen rendering.

Any thoughts on that, or is that behavior really wanted? It's really limiting for applications that could be much faster

reconbot, 2017-09-24:
5x and 6x have very different windows read/write approaches. Windows and Unix also have very different approaches in general. And windows does not support `uv_poll`. I'm going to talk about unix, but if this is an issue on windows too then maybe the problem is at a different level of abstraction.

I'm pretty sure that even if we're limited to a vsync number of events the data will buffer in the OS. I don't know what the limits of this buffer is or when it would impact data flow. 

We also keep reading if there is data available, [we only stop if there is none left](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/lib/bindings/unix-read.js#L13-L27) or our buffers are full. In the earlier case we use `uv_poll` to tell us there's more data. In the later case, we wait for data to be read and then keep reading.

With writing we keep writing until the OS asks us to stop, and then we wait for the FD to be writable again via `uv_poll`. So this should be similar to reads.

>It seems to me, that it should be using some sort of socket piping when reading from the Serial, but that seems unclear as no one has done that before.

I don't understand what you mean here. We treat the serial port basically the same as a file with regards to reading and writing.

> Any thoughts on that, or is that behavior really wanted? It's really limiting for applications that could be much faster

It's definitely a good time to work on improving this. The 6x beta does a very good improvement for windows and there's bound to be much opportunity throughout the library.

reconbot, 2017-10-07:
Have you been able to confirm anything here?

ivanseidel, 2017-10-07:
You can confirm it, by simply logging `Date.now() - lastDate` in `lib/bindings/poller.js` in `poll(eventFlag)` function.

That is the method called to check if data has been received...

Pooling is done in order to verify changes, while it would be best to use interrupts to trigger readings. But I'm not sure how that could be done.

I tested in both Windows and Mac, and both gave me a exact 16ms period, and Data always snaps to that while reading...

reconbot, 2017-10-07:
We don't actually poll, it's epoll under the hood for Linux and readfileex
for Windows. Might be worth opening an issue with libuv.

reconbot, 2017-10-07:
It's technically uv_poll for Linux and osx ftr

cinderblock, 2018-03-26:
Are there any tricks that could be used to reduce the latency from when the bytes come in on the wire to when they're available as `port.on('data', ...)`?

I have ~500Hz data coming in to the machine from a sensor. Most of the time, it works fine but for 1~5% of messages, they all arrive basically at once.

Crazy thought is maybe it would be possible to integrate the readline parser and notifying the javascript as soon as a `\n` byte is received. Or maybe not and it's all the OS that is off doing something else and there would be nothing we can do in userland.

This is happening on Windows and Linux.

I could go to a realtime system but it would be nice to stay in the node world.

closed 2018-12-15 by reconbot