serialport/node-serialport #1346: cant't get npmlog?

frank-dspeed, 2017-09-24:
```
> serialport@6.0.0-beta2 install /srv/drivers/cctalk-devices/node_modules/serialport
> prebuild-install || node-gyp rebuild

module.js:491
    throw err;
    ^

Error: Cannot find module 'npmlog'
    at Function.Module._resolveFilename (module.js:489:15)
    at Function.Module._load (module.js:439:25)
    at Module.require (module.js:517:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/srv/drivers/cctalk-devices/node_modules/prebuild-install/bin.js:4:11)
    at Module._compile (module.js:573:30)
    at Object.Module._extensions..js (module.js:584:10)
    at Module.load (module.js:507:32)
    at tryModuleLoad (module.js:470:12)
    at Function.Module._load (module.js:462:3)
make: Verzeichnis „/srv/drivers/cctalk-devices/node_modules/serialport/build“ wird betreten
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Verzeichnis „/srv/drivers/cctalk-devices/node_modules/serialport/build“ wird verlassen

```

reconbot, 2017-09-24:
What version of node? it's a dep of prebuild

reconbot, 2017-10-07:
I assume this is no longer an issue? I'll close for now.

closed 2017-10-07 by reconbot

frank-dspeed, 2017-10-09:
@reconbot it could not be reproduced it happend after npm install 5.x over 4.x it can be closed