serialport/node-serialport #1349: Webpack and Electron: Uncaught ReferenceError: bindings is not defined

patrykkalinowski, 2017-09-26:
- __SerialPort Version__: 5.0.0
- __NodeJS Version__: v8.2.1
- __Operating System__ and __Hardware Platform__: OSX, Macbook Pro

### Summary of Problem
I'm trying to use SerialPort in Electron using webpack. I have installed SerialPort for Electron and it works, the problem is when introducing webpack.

Browser console returns:
```
Uncaught ReferenceError: bindings is not defined
    at Object.<anonymous> (bundle.js:13720)
    at __webpack_require__ (bundle.js:20)
    at Object.<anonymous> (bundle.js:73620)
    at __webpack_require__ (bundle.js:20)
    at Object.<anonymous> (bundle.js:73401)
    at Object.<anonymous> (bundle.js:73404)
    at __webpack_require__ (bundle.js:20)
    at Object.<anonymous> (bundle.js:71010)
    at __webpack_require__ (bundle.js:20)
    at Object.<anonymous> (bundle.js:70981)
```

### Steps and Code to Reproduce the Issue

At first, I had following errors during compilation:
```
WARNING in ./node_modules/bindings/bindings.js
81:22-40 Critical dependency: the request of a dependency is an expression

WARNING in ./node_modules/bindings/bindings.js
81:43-53 Critical dependency: the request of a dependency is an expression

ERROR in ./node_modules/serialport/lib/bindings/linux-list.js
Module not found: Error: Can't resolve 'child_process' in '/Users/patryk/Code/app/node_modules/serialport/lib/bindings'
 @ ./node_modules/serialport/lib/bindings/linux-list.js 3:21-45
 @ ./node_modules/serialport/lib/bindings/linux.js
 @ ./node_modules/serialport/lib/bindings/auto-detect.js
 @ ./node_modules/serialport/lib/index.js
 @ ./src/js/serialport.js
 @ ./src/js/app.js
 @ multi (webpack)-dev-server/client?http://localhost:8081 ./src/js/app.js

ERROR in ./node_modules/bindings/bindings.js
Module not found: Error: Can't resolve 'fs' in '/Users/patryk/Code/app/node_modules/bindings'
 @ ./node_modules/bindings/bindings.js 6:9-22
 @ ./node_modules/serialport/lib/bindings/linux.js
 @ ./node_modules/serialport/lib/bindings/auto-detect.js
 @ ./node_modules/serialport/lib/index.js
 @ ./src/js/serialport.js
 @ ./src/js/app.js
 @ multi (webpack)-dev-server/client?http://localhost:8081 ./src/js/app.js

ERROR in ./node_modules/serialport/lib/bindings/unix-read.js
Module not found: Error: Can't resolve 'fs' in '/Users/patryk/Code/app/node_modules/serialport/lib/bindings'
 @ ./node_modules/serialport/lib/bindings/unix-read.js 2:11-24
 @ ./node_modules/serialport/lib/bindings/linux.js
 @ ./node_modules/serialport/lib/bindings/auto-detect.js
 @ ./node_modules/serialport/lib/index.js
 @ ./src/js/serialport.js
 @ ./src/js/app.js
 @ multi (webpack)-dev-server/client?http://localhost:8081 ./src/js/app.js

ERROR in ./node_modules/serialport/lib/bindings/unix-write.js
Module not found: Error: Can't resolve 'fs' in '/Users/patryk/Code/balloon-navigator/node_modules/serialport/lib/bindings'
 @ ./node_modules/serialport/lib/bindings/unix-write.js 2:11-24
 @ ./node_modules/serialport/lib/bindings/linux.js
 @ ./node_modules/serialport/lib/bindings/auto-detect.js
 @ ./node_modules/serialport/lib/index.js
 @ ./src/js/serialport.js
 @ ./src/js/app.js
 @ multi (webpack)-dev-server/client?http://localhost:8081 ./src/js/app.js
webpack: Failed to compile.
```
I fixed it by adding to `webpack.config.js`:
```
externals: {
        fs: 'fs',
        child_process: 'child_process',
        bindings: 'bindings'
    }
```
Now webpack compiles application without errors, but console shows me following error:
```
Uncaught ReferenceError: bindings is not defined
    at Object.<anonymous> (bundle.js:13720)
    at __webpack_require__ (bundle.js:20)
    at Object.<anonymous> (bundle.js:73620)
    at __webpack_require__ (bundle.js:20)
    at Object.<anonymous> (bundle.js:73401)
    at Object.<anonymous> (bundle.js:73404)
    at __webpack_require__ (bundle.js:20)
    at Object.<anonymous> (bundle.js:71010)
    at __webpack_require__ (bundle.js:20)
    at Object.<anonymous> (bundle.js:70981)
```

Adding variables *doesn't help*:
```
var bindings = require('binding');
var fs = require('fs');
var child_process = require('child_process');
```

Any ideas?

reconbot, 2017-09-26:
Webpack is not supported, it can be made to work but you have to provide
the binary module as well. It's much easier to make serialport an external
module.

As for the fs errors I can't explain that fs is a node internal module.

patrykkalinowski, 2017-09-27:
Thanks @reconbot. Looks like defining serialport as external module in webpack partly did the trick:
```
externals: {
        serialport: "serialport"
    }
```

Browser error is different now:
```
Uncaught ReferenceError: serialport is not defined
    at Object.<anonymous> (bundle.js:65659)
    at __webpack_require__ (bundle.js:20)
    at Object.module.exports (bundle.js:65629)
    at __webpack_require__ (bundle.js:20)
    at Object.<anonymous> (bundle.js:31520)
    at __webpack_require__ (bundle.js:20)
    at Object.<anonymous> (bundle.js:25595)
    at __webpack_require__ (bundle.js:20)
    at _ol_ (bundle.js:63)
    at bundle.js:66
```
I suppose this is strictly webpack related problem, so I'll try to seek more help there.

patrykkalinowski, 2017-09-27:
OK, so I described my problem in detail on [Stack Overflow](https://stackoverflow.com/questions/46442623/node-serial-port-as-external-module-in-webpack-module-not-found)

My problem is, I don't know how to include serialport binary inside JS app. `require('serialport');` returns `serialport is not defined`. No idea what to do now.

reconbot, 2017-09-27:
You can copy it next to your bundle in the node_modules folder.

I'm sorry, webpack is such a can of worms I can't really tell you more than that. There's 1000 ways it can go wrong with binary modules and if you search the closed issues you'll find a lot of people who got it to work and 1000 ways they didn't.

closed 2017-09-27 by reconbot