serialport/node-serialport #1350: Windows7 64x CPU accounted for 100%

grubed, 2017-09-27:
- __SerialPort Version__:5.0.0 -- 6.0.0 
- __NodeJS Version__:8.2.1
- __Operating System__ and __Hardware Platform__:Windows7 64

- Have you checked the right __version__ of the __api docs?__:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?:

### Summary of Problem
CPU accounted for 100%

### Steps and Code to Reproduce the Issue

reconbot, 2017-09-27:
Do you have any sample code? This should be vast improved with the v6 betas.

grubed, 2017-09-27:
const serialport = require('serialport');

  const port = new serialport('COM3', {
    baudRate: 9600
  });

  port.on('open', () => console.log('Port open'));

  port.on('close', () => console.log('Port close'));
  port.on('error', (err) => console.log('Port error'));

  port.on('data', (data) => {
  }

reconbot, 2017-09-27:
And you notice no difference between 5x and 6x?

grubed, 2017-09-28:
today i use "serialport": "^4.0.7"，cpu 0% Normal ！

marliotto, 2017-09-28:
I have same problem with serialport 5.0.0.
Node.js 7.4.0
Windows 7 x64
Electron 1.6.13

reconbot, 2017-10-07:
What about 6.0?

reconbot, 2017-10-07:
We released serialport@6.0.0-beta3 today. 6x should solve your problems.

grubed, 2017-10-09:
use 6.0.0-beta3 , then Port open and program exit, can not loop

grubed, 2017-10-09:
i use node app.js for start

reconbot, 2017-10-20:
v6.0.1 fixes some event loop bugs, please give it a try

grubed, 2017-10-20:
same to 6.0.0-beta3 in Node.js 8.6.0 Windows 7 x64

reconbot, 2017-10-20:
6.0.2 is now in beta

grubed, 2017-10-24:
v6.0.3 loop bugs in Node.js 8.6.0 Windows 7 x64

reconbot, 2017-10-26:
v6.0.4 has been released which fixes those bugs

grubed, 2017-10-30:
yes v6.0.4 it is ok

closed 2017-10-30 by grubed