serialport/node-serialport #1353: Error while opening serial port

gozzoo, 2017-10-07:
- __SerialPort Version__: 6.0.0-beta2
- __NodeJS Version__: 7.9.0 (Electron 1.8.1)
- __Operating System__ and __Hardware Platform__: Windows 7

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: yes

### Summary of Problem
I get this error when trying to open existing but not connected port:

> Error: Open (SetCommState): Unknown error code 87

### Steps and Code to Reproduce the Issue

Here is my code: 

```
let port = 'COM1'
let conf = { baudRate: Number(speed) }
port = new SerialPort(port, conf)
const parser = new Readline();
port.pipe(parser)
parser.on('data', (data) => {
  console.log(data)
}) 
port.on('error', err =>  console.error(t.port + ' error: ', err))
```

reconbot, 2017-10-07:
Does it work in node?

gozzoo, 2017-10-07:
It works fine with latestd node 8.6.0 (latest). I'll downgrade node to the version of Electron and I'll try again.

gozzoo, 2017-10-07:
It also works fine with node 7.9.0

What's interesting tough is that witn serialport 5.0.0 my program hangs, and with 6.0.0-beta2 it exists immediately. I don't know if this is related to the error I'm getting  with Electron.

gozzoo, 2017-10-07:
I reproduced the problem with all node versions. I was pasing 0 as `baudRate`. I don't know if 0 is a valid value but 'error code 87' is not a very clear error message.

reconbot, 2017-10-07:
It appears that the c++ code is running ok, the open error `SetCommState` has to do with configuring the ports setting and baudrate... A baudrate of 0 is not valid however.

reconbot, 2017-10-07:
It's what the OS gives us ¯\_(ツ)_/¯

reconbot, 2017-10-07:
I would 100% take a pr that throws on this error, if it hit you it has hit
others too.

reconbot, 2017-10-08:
Thanks for the patch @gozzoo!

closed 2017-10-08 by reconbot