serialport/node-serialport #1355: some question about the releases page

hanfengcan, 2017-10-09:
I found that https://github.com/EmergingTechnologyAdvisors/node-serialport/releases has some .gz file

what the meaning such as `electron-v47`. 

v47 is the version? but the electron vesion is 1.xxx

What is the use of these?

reconbot, 2017-10-09:
They will auto download with "prebuild" the version number is the node abi
version.

closed 2017-10-09 by reconbot

hanfengcan, 2017-10-10:
@reconbot how can I use this prebuild? 
but the way, have a easy way to use serialport under electron?
I use a new NoteBook, I find that is difficulty  to rebuild the serialport

reconbot, 2017-10-10:
you need to use electron rebuild, there are instructions in the readme https://github.com/EmergingTechnologyAdvisors/node-serialport#electron

hanfengcan, 2017-10-10:
@reconbot i now :cry: