serialport/node-serialport #1356: Port hangs on polling for readable w/strings vs chunks

petertorelli, 2017-10-09:
- __SerialPort Version__: 5.0.0
- __NodeJS Version__: v7.7.2
- __Operating System__ and __Hardware Platform__: Darwin Kernel Version 16.7.0, 2013 MacBook PRO

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem

I have an STMicoro L476RG board set up as a simple command parser. The parser just responds to a handful of recognized strings that end with '%'. When I issue a command more than 7 bytes, the device stops responding, and I have to unplug/replug the USB. 

This hang only happens with the below code. It does not happen with `serialport-term`, MiniCom, TeraTerm (on Windows) or 'screen': I can send as many bytes as a want until the firmware truncates the incoming string.

Turning on `DEBUG=*`, I see it enters polling for readable, but then never comes out of that state on the failure:

~~~
  serialport:main _read reading +0ms
  serialport:bindings read +0ms
  serialport:unixRead Starting read +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +0ms
  *serialport:poller Polling for "readable" +0ms*
mmmmmmmm%
  serialport:main _write 9 bytes of data +496ms
  serialport:bindings write 9 bytes +0ms
  serialport:unixWrite Starting write 9 bytes offset 0 bytesToWrite 9 +0ms
  serialport:unixWrite write returned null 9 +0ms
  serialport:unixWrite wrote 9 bytes +0ms
  serialport:unixWrite Finished writing 9 bytes +0ms
  serialport:main binding.write write finished +0ms
  serialport:main drain +0ms
  serialport:bindings drain +0ms
  serialport:main binding.drain finished +1ms
drained
mmm%
  serialport:main _write 4 bytes of data +501ms
  serialport:bindings write 4 bytes +0ms
~~~

I thought it was my firmware until I noticed it works fine on three other terminal programs. It also works fine using the serialport-term.

The only difference I can see between my implementation and the provided `serialport-term` is that the latter uses raw mode chunks and I am passing strings.

### Steps and Code to Reproduce the Issue

I can't expect you to have the same firmware, so I'm not sure how to proceed, but perhaps this sample code that I tested with will help?

~~~
#!/usr/bin/env node
'use strict';
let portName = '/dev/cu.usbmodem14143';
const serialPort = require('serialport');
let port;
let commandTable = [
	'mmm%',
	'mmmm%',
	'mmmmm%',
	'mmmmmm%',
	'mmmmmmm%', // this one succeeds
	'mmmmmmmm%', // dies here, but serialport-term, MiniCom, TerrTerm and 'screen' work fine if I manually TYPE in these strings
];
let cmdIdx = 0;
port = new serialPort(portName, {
	baudRate: 115200,
}).on('open', () => {
	console.log(`[${portName}]: open`);
	setInterval(() => {
		console.log(commandTable[cmdIdx]);
		port.write(commandTable[cmdIdx], err => {
			if (err) {
				console.log('Port write error:', err);
				port.close();
				process.exit();
			}
			port.drain(() => {
				console.log("drained");
			});
		});
		cmdIdx = (cmdIdx < commandTable.length - 1) ? cmdIdx + 1 : 0;
	},
	500);
}).on('data', data => {
	console.log(`[${portName}]: data: ${data}`);
}).on('error', err => {
	console.log(`[${portName}]: ${err}`);
}).on('disconnect', () => {
	console.log(`[${portName}]: disconnect`);
	process.exit(1);
});
~~~

reconbot, 2017-10-09:
I think strings vs buggers is a red herring. You can test that by calling `Buffer.from(string)` on your commands before sending them.

What I'm seeing from your logs is that the second write never starts. The `serialport:unixWrite` should immediately follow if there is no input errors. This all happens in JS and shouldn't have anything to do with what you're writing to.

A few notes:
- The disconnect event no longer exists
- You can call drain after calling write without waiting for the write callback
- Write errors will trigger the port error event
- You can start writing whenever you like, it will keep the data queued until the port is open.

I changed your code a bit to start writing after the drain returns.

```js
const port = new SerialPort(portName, {
  baudRate: 115200,
}).on('open', () => {
  console.log(`[${portName}]: open`)
}).on('data', data => {
  console.log(`[${portName}]: data: ${data}`)
}).on('error', err => {
  console.log(`[${portName}]: ${err}`)
  process.exit()
})

let cmdIdx = 0

function writeAndDrain(data, cb) {
  port.write(data)
  port.drain(cb)
}

function writeNextCommand() {
  const nextCommand = commandTable[cmdIdx]
  console.log(`Writing ${nextCommand}`)
  writeAndDrain(nextCommand, () => {
    cmdIdx = (cmdIdx < commandTable.length - 1) ? cmdIdx + 1 : 0
    setTimeout(writeNextCommand, 500)
  })
}
writeNextCommand()
```

petertorelli, 2017-10-10:
@reconbot Thanks for taking the time to reply. It occurred to me that manual entry on serialport-term and minicom is passing because there is a delay between keystrokes. I simply cut and pasted a long string of commands into those apps and the same crash happens so it is clearly the firmware.

Thanks. Closing this.

closed 2017-10-10 by petertorelli