serialport/node-serialport #1357: Electron crashesh under Windows when reading from serialport and sending data with ipc

gozzoo, 2017-10-10:
- __SerialPort Version__: 6.0.0-beta3
- __NodeJS Version__: Electron 1.8.1
- __Operating System__ and __Hardware Platform__: Windows 10 32bit

### Summary of Problem
After I start a very simple Electron app which gets data from the serial port and sends it using IPC to the browser windows the application freezes and becomes unresponsive, then Windows shows message 'Electron has stopped working'. 

If I send random data to the window everything is working fine. Only if the data comes from the serial port then I can see crashes.

If I just read the data from the serial port and print it on the console everything is fine. It seems that both things together are related to the crash - reading data from the port and sending it with IPC.

### Steps and Code to Reproduce the Issue

Here is my Electron app.

main.js
```

var electron = require('electron');
var app = electron.app;
var BrowserWindow = electron.BrowserWindow;
const path = require('path')
const url = require('url')

var win = null;

app.on('ready', function(){
	win = new BrowserWindow({
		width: 800,
		height: 600
	})

	win.loadURL(url.format({
    pathname: path.join(__dirname, 'index.html'),
    protocol: 'file:',
    slashes: true
  }))
	win.webContents.openDevTools()
})

var SerialPort = require('serialport')
const port = new SerialPort('COM4', {baudRate: 2400})
const Readline = SerialPort.parsers.Readline
const parser = new Readline();
port.pipe(parser);

parser.on('data', data =>  win.send('serial', data));
port.on('error', error => console.log('  [ERROR]  ' + error))
```

index.html

```
<!DOCTYPE html>
<html>
  <head>
    <title>Hello World!</title>
  </head>
  <body>
    <h1>Hello World!</h1>
    We are using io.js <script>document.write(process.version)</script>
    and Electron <script>document.write(process.versions['electron'])</script>.

    <div id="dbrows">
      <h2>Database Rows</h2>
    </div>
    <script>
      const {ipcRenderer} = require('electron')
      ipcRenderer.send('serial', 'ping')
      ipcRenderer.on('serial', (event, arg) => {
        console.log(arg) 
      })
    </script>
  </body>
</html>
```

reconbot, 2017-10-10:
This sounds not like a serialport problem but an electron issue. You have an ipc problem? You can you provide any details about the crash? Stack trace? 

Looking at the docs, you probably need to turn your data into an array or a string if you want to send it via ipc. Serialport provides buffers. `Array.from(data)` and `data.toString('binary')` are both good ways.

closed 2017-10-10 by reconbot

gozzoo, 2017-10-10:
> This  sounds not like a serialport problem but an electron issue. You have an ipc problem?

It is definitely related to the serialport module. When I exclude all serial port code everything is working fine. I just noticed that if there is any ipc communication going on while reading data from a serial port also causes a crash, like this.

```
port.on('data', data =>  console.log(data));
setInterval(_ => win.webContents.send('serial', 'test'), 100)
```
If I comment out any of these lines there is no crash.
> Looking at the docs, you probably need to turn your data into an array or a string if you want to send it via ipc. Serialport provides buffers. Array.from(data) and data.toString('binary') are both good ways.

`parser.on('data', ...)` gives me a string.

gozzoo, 2017-10-10:
Here is the stacktrace:

```
ntdll.dll!774db68a()	Unknown
[Frames below may be incorrect and/or missing, no symbols loaded for ntdll.dll]	
electron.exe!views::NativeViewHost::Layout() Line 98
electron.exe!scoped_refptr<ui::LayerAnimator>::Release(ui::LayerAnimator * ptr) Line 573
electron.exe!scoped_refptr<ui::LayerAnimator>::operator=(ui::LayerAnimator * p) Line 495
electron.exe!ui::Layer::SetAnimator(ui::LayerAnimator * animator) Line 311
electron.exe!ui::Layer::~Layer() Line 142
[External Code]	
electron.exe!content::DelegatedFrameHost::UpdateGutters() Line 280
electron.exe!content::DelegatedFrameHost::WasResized() Line 260
electron.exe!content::RenderWidgetHostViewAura::InternalSetBounds(const gfx::Rect & rect) Line 2138
electron.exe!content::RenderWidgetHostViewAura::SetSize(const gfx::Size & size) Line 581
electron.exe!content::RenderWidgetHostViewAura::OnBoundsChanged(const gfx::Rect & old_bounds, const gfx::Rect & new_bounds) Line 1498
electron.exe!aura::Window::OnLayerBoundsChanged(const gfx::Rect & old_bounds) Line 1021
electron.exe!ui::Layer::SetBoundsFromAnimation(const gfx::Rect & bounds) Line 1008
electron.exe!ui::LayerAnimator::SetBounds(const gfx::Rect & value) Line 112
electron.exe!aura::Window::SetBoundsInternal(const gfx::Rect & new_bounds) Line 689
electron.exe!aura::Window::SetBounds(const gfx::Rect & new_bounds) Line 302
electron.exe!content::RenderWidgetHostViewAura::InternalSetBounds(const gfx::Rect & rect) Line 2134
electron.exe!content::RenderWidgetHostViewAura::SetSize(const gfx::Size & size) Line 581
electron.exe!content::WebContentsViewAura::OnBoundsChanged(const gfx::Rect & old_bounds, const gfx::Rect & new_bounds) Line 1087
electron.exe!aura::Window::OnLayerBoundsChanged(const gfx::Rect & old_bounds) Line 1021
electron.exe!ui::Layer::SetBoundsFromAnimation(const gfx::Rect & bounds) Line 1008
electron.exe!ui::LayerAnimator::SetBounds(const gfx::Rect & value) Line 112
electron.exe!aura::Window::SetBoundsInternal(const gfx::Rect & new_bounds) Line 689
electron.exe!aura::Window::SetBounds(const gfx::Rect & new_bounds) Line 302
electron.exe!views::NativeViewHostAura::ShowWidget(int x, int y, int w, int h) Line 164
electron.exe!views::NativeViewHost::Layout() Line 98
electron.exe!views::View::BoundsChanged(const gfx::Rect & previous_bounds) Line 2178
electron.exe!views::View::SetBoundsRect(const gfx::Rect & bounds) Line 348
electron.exe!views::WebView::OnBoundsChanged(const gfx::Rect & previous_bounds) Line 141
electron.exe!views::View::BoundsChanged(const gfx::Rect & previous_bounds) Line 2173
electron.exe!views::View::SetBoundsRect(const gfx::Rect & bounds) Line 348
electron.exe!brightray::InspectableWebContentsViewViews::Layout() Line 225
electron.exe!views::View::BoundsChanged(const gfx::Rect & previous_bounds) Line 2178
electron.exe!views::View::SetBoundsRect(const gfx::Rect & bounds) Line 348
electron.exe!atom::NativeWindowViews::Layout() Line 1368
electron.exe!views::View::BoundsChanged(const gfx::Rect & previous_bounds) Line 2178
electron.exe!views::View::SetBoundsRect(const gfx::Rect & bounds) Line 348
electron.exe!views::View::SetBounds(int x, int y, int width, int height) Line 325
electron.exe!views::ClientView::Layout() Line 71
electron.exe!views::View::BoundsChanged(const gfx::Rect & previous_bounds) Line 2178
electron.exe!views::View::SetBoundsRect(const gfx::Rect & bounds) Line 348
electron.exe!views::NonClientView::Layout() Line 178
electron.exe!views::View::BoundsChanged(const gfx::Rect & previous_bounds) Line 2178
electron.exe!views::View::SetBoundsRect(const gfx::Rect & bounds) Line 348
electron.exe!views::FillLayout::Layout(views::View * host) Line 19
electron.exe!views::View::Layout() Line 628
electron.exe!views::View::BoundsChanged(const gfx::Rect & previous_bounds) Line 2178
electron.exe!views::View::SetBoundsRect(const gfx::Rect & bounds) Line 348
electron.exe!views::View::SetBounds(int x, int y, int width, int height) Line 325
electron.exe!views::View::SetSize(const gfx::Size & size) Line 354
electron.exe!views::Widget::OnNativeWidgetSizeChanged(const gfx::Size & new_size) Line 1123
electron.exe!views::DesktopNativeWidgetAura::OnHostResized(const aura::WindowTreeHost * host) Line 1173
electron.exe!aura::WindowTreeHost::OnHostResizedInPixels(const gfx::Size & new_size_in_pixels) Line 303
electron.exe!views::HWNDMessageHandler::ClientAreaSizeChanged() Line 1147
electron.exe!views::HWNDMessageHandler::OnWindowPosChanged(tagWINDOWPOS * window_pos) Line 2517
electron.exe!views::HWNDMessageHandler::_ProcessWindowMessage(HWND__ * hWnd, unsigned int uMsg, unsigned int wParam, long lParam, long & lResult, unsigned long dwMsgMapID) Line 440
electron.exe!views::HWNDMessageHandler::OnWndProc(unsigned int message, unsigned int w_param, long l_param) Line 915
electron.exe!gfx::WindowImpl::WndProc(HWND__ * hwnd, unsigned int message, unsigned int w_param, long l_param) Line 303
electron.exe!base::win::WrappedWindowProc<&gfx::WindowImpl::WndProc>(HWND__ * hwnd, unsigned int message, unsigned int wparam, long lparam) Line 76
[External Code]	
electron.exe!views::HWNDMessageHandler::OnSysCommand(unsigned int notification_code, const gfx::Point & point) Line 2286
electron.exe!views::HWNDMessageHandler::_ProcessWindowMessage(HWND__ * hWnd, unsigned int uMsg, unsigned int wParam, long lParam, long & lResult, unsigned long dwMsgMapID) Line 437
electron.exe!views::HWNDMessageHandler::OnWndProc(unsigned int message, unsigned int w_param, long l_param) Line 915
[... cut ...]
electron.exe!content::RunNamedProcessTypeMain(const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & process_type, const content::MainFunctionParams & main_function_params, content::ContentMainDelegate * delegate) Line 438
electron.exe!content::ContentMainRunnerImpl::Run() Line 740
electron.exe!service_manager::Main(const service_manager::MainParams & params) Line 179
electron.exe!content::ContentMain(const content::ContentMainParams & params) Line 19
electron.exe!wWinMain(HINSTANCE__ * instance, HINSTANCE__ * __formal, wchar_t * cmd, int __formal) Line 159
[External Code]	

```

reconbot, 2017-10-10:
I'll watch the electron issue, I'm not equipped to debug electron like they are, happy to work with them if it does turn out to be a direct serialport issue.