serialport/node-serialport #1359: Serial port stops emmiting 'data' event with Electron under Windows

gozzoo, 2017-10-10:
- __SerialPort Version__: 6.0.0-beta3
- __NodeJS Version__: Electron 1.8.1
- __Operating System__ and __Hardware Platform__: Windows 10 32bit

### Summary of Problem

I have a very simple Electron app which gets data from the serial port and displays it in the browser window. After I leave it for a while (~5min) the updates stop.

### Steps and Code to Reproduce the Issue
Here is the code:

main.js
```
var electron = require('electron');
var app = electron.app;
var BrowserWindow = electron.BrowserWindow;

var win = null;
app.on('ready', function(){
  win = new BrowserWindow({width: 800, height: 600})
  win.loadURL('file:///index.html')
})
```

index.html
```
<!DOCTYPE html>
<html>
  <body>
    <div id="msg">abc</div>
    <script>
      var e = document.getElementById("msg");
      var SerialPort = require('serialport')
      const port = new SerialPort('COM4', {baudRate: 2400})
      const Readline = SerialPort.parsers.Readline
      const parser = new Readline();
      port.pipe(parser);
      var i = 0;
      parser.on('data', data => e.innerHTML = (data + ' ' + i++))
    </script>
  </body>
</html>
```

reconbot, 2017-10-11:
Are there errors? Where does the data come from?

gozzoo, 2017-10-11:
No, there are no errors. The data comes from a hardware sensor which has been working for years with many kinds of other software. The communication is unidirectional, it just sends data.
To eleminate other possible causes factors that can cause for this problem (hardrware, connection, driver, OS) I developed a test example that does the same thing in C# and it keep reading the data forever with no glitches.

reconbot, 2017-10-11:
What about node without electron?

Also please check the readme about debug mode and share that out put.

gozzoo, 2017-10-12:
It works fine with node. There is also no problem if I run it with Electron in the main process, if there is nothing else going on there. As reported in #1357 if there is IPC commnication while reading from serial port Electron crashes.

I'll test more in debug mode.

reconbot, 2017-10-14:
If it works with node then I'm not sure what I can do. If the electron people can help point to anything we can change I'm for it.

closed 2017-10-14 by reconbot

gozzoo, 2017-10-20:
I don't have this problem any more with release 6.0.1.

reconbot, 2017-10-20:
Great!

aguegu, 2017-12-22:
@gozzoo you mean serialport v6.0.1 fixs your problem? 

I do not have the luck you got. I have a serial device doing 'echo' program. The `serialport.on('data')` event trigger randomly, only receive like 20% of all the data...

But if I ran the code without electron, it works fine.