serialport/node-serialport #1360: Congrats on the 6.0.0 release... you forgot to prefix tag with v

MylesBorins, 2017-10-11:
oh hey.

All previous releases have been prefix'd with a `v`... so this broke citgm :(

Let us know if you plan to drop the `v` moving forward and we can update the lookup

reconbot, 2017-10-11:
I switched to prebuild. It doesn't use a v, so I don't have a v anymore.
¯\_(ツ)_/¯

reconbot, 2017-10-18:
I can guarantee you that it will continue to have a `v` for quite some time.

reconbot, 2017-10-18:
I can guarantee you that it will continue to not have a `v` for quite some time.

closed 2017-10-18 by reconbot

MylesBorins, 2017-10-18:
LOL I WAS BACKWARDS

you now have a v... didn't have a v before

reconbot, 2017-10-18:
I totally had it right and then changed it to fix your message :dies:

https://github.com/EmergingTechnologyAdvisors/node-serialport/tree/v6.0.1 

Keeping the V

MylesBorins, 2017-10-18:
Lololol