serialport/node-serialport #1362: Serialport not able to connect at 250000 baudrate to Ultimaker

matthova, 2017-10-12:
- __SerialPort Version__: 6
- __NodeJS Version__: 8.7
- __Operating System__ and __Hardware Platform__: Raspbian Jessie Lite on Raspberry pi

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? N/A
- Are you using __Electron__ and have you checked the _Electron Docs_?: N/A

### Summary of Problem
Unable to connect to an Ultimaker at 250000 baudrate. Identical code runs great on 2014 Mac running OSX 10.12
### Steps and Code to Reproduce the Issue
```
const serialport = require('serialport');
const port = new serialport('/dev/ttyACM0', { baudRate: 250000 });
port.on('data', (data) => { console.log('data', data.toString()); });
port.on('open', () => {
  setTimeout(() => {
    console.log('writing to port');
    port.write('M115\n');
  }, 5000);
  setTimeout(() => {
    console.log('writing to port');
    port.write('M115\n');
  }, 10000);
});
```

matthova, 2017-10-13:
I've created a repo for reproducing this bug:
https://github.com/matthova/serialport-6-250000-test

matthova, 2017-10-13:
Just attempted the same code on Raspbian Stretch lite with the same failure mode of no reply.

fivdi, 2017-10-15:
Linux has standard baud rates and custom baud rates. Standard baud rates are handled by serialport and can be seen [here](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/v6.0.0/src/serialport_unix.cpp#L31-L68). serialport also has code for handling custom baud rates [here](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/v6.0.0/src/serialport_unix.cpp#L111-L133).

250000 is a custom baud rate and for some reason it doesn't work. I tried out the code in your repo and attempting to set the baud rate to 250000 actually sets it to 38400. I'm not sure if this is a serialport issue, a Raspbian driver issue or something else. Ubuntu 14.04 also has the issue.

What does work is if the baud rate is set to a standard baud rate like 230400 which is fairly close to 250000.

matthova, 2017-10-19:
@fivdi thanks for confirming the issue and for more background on where it's handled. I wish I could just set the baudrate to 250000 but I'm stuck working with 3rd party hardware that is set at 250000. @reconbot let me know if there is anything I can do to dig into the code and help with a solution.

reconbot, 2017-10-20:
The c++ code needs to use the `termios2` struct on linux to get that rate. I tried a while back without success (I knew even less what I was doing back then). The branch is even still around (though horribly out of date) https://github.com/EmergingTechnologyAdvisors/node-serialport/compare/baudrates?expand=1

closed 2018-11-22 by stale[bot]