serialport/node-serialport #1363: Node event loop stops while the write operation is in progress

kimushu, 2017-10-16:
- __SerialPort Version__: 6.0.0
- __NodeJS Version__: 6.9.5 (64-bit)
- __Operating System__ and __Hardware Platform__: Windows 10 Pro (10.0.15063) x64
  - Serial port: FTDI USB-Serial converter cable TTL-232R-3V3 (FT232RQ chip)

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? no
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem
Event loop stops even if there is an incomplete write operation.
For example, simple open & write program ([here](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/v6.0.0/README.md#opening-a-port)) does not work.
I guess the new I/O for windows (#1221) does not preserve any AsyncWorker in write operation.

### Steps and Code to Reproduce the Issue
- Event loop stops before `message written` message if there are no pending callbacks nor timers.
```
$ cat open_write.js
var SerialPort = require('serialport');
var port = new SerialPort('COM8');

port.write('main screen turn on', function(err) {
  if (err) {
    return console.log('Error on write: ', err.message);
  }
  console.log('message written');
});

// Open errors will be emitted as an error event
port.on('error', function(err) {
  console.log('Error: ', err.message);
});

$ node open_write.js
                                        # <- No output here!
$ echo $?
0                                       # <- node stopped without error...
```
- If there are some pending callbacks or timers, write operation & callback seem to be ok. However, event loop does not stop even if write operation has been finished
```
$ cat open_write_with_timer.js
var SerialPort = require('serialport');
var port = new SerialPort('COM8');

port.write('main screen turn on', function(err) {
  if (err) {
    return console.log('Error on write: ', err.message);
  }
  console.log('message written');
});

// Open errors will be emitted as an error event
port.on('error', function(err) {
  console.log('Error: ', err.message);
});

setTimeout(() => { console.log("workaround!"); }, 1000);

$ node open_write_with_timer.js
message written                         # <- Write callback is ok...
workaround!
                                        # However, event loop seems be frozen (Node is still running...)
```

reconbot, 2017-10-16:
So if we have a pending write the event loop still appears empty and node will quit?

kimushu, 2017-10-17:
Correct. There are two problems.
As for the former one (node quit during write), I have an idea to fix it by adding uv_idle (https://github.com/kimushu/node-serialport/commit/f3a94bf1a3e136c72e329dc5b5c5b102160b90c1)

kimushu, 2017-10-18:
uv_async_t handle is not correctly closed ([here](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/v6.0.1/src/serialport_win.cpp#L353)). This disturbs finish of event loop.
I have a proposal to solve both problems in this issue. See #1367.

reconbot, 2017-10-18:
This is was great, thank you so much. It's always nice when people give good bug reports, and it's even more amazing when they close them too. =)

kimushu, 2017-10-19:
It's a pleasure to contribute :-)

closed 2017-10-26 by reconbot