serialport/node-serialport #1365: node_modules_version

peiyc, 2017-10-17:
- __SerialPort Version__:6.0.0
- __NodeJS Version__:7.9.0  8.0.0
- __Operating System__ and __Hardware Platform__:

- Have you checked the right __version__ of the __api docs?__:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?:

I used serialport with electron(1.7.9).
this is the error: 
caught Error: The module '/Users/web/Documents/practice/electron-quick-start/node_modules/serialport/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57.(node 8.0.0+) This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

the problem is node_module_version.
the node_module version in node(7.9.0) is 51,
but in node(8.0.0) is 57,
I don`t know how to use the version 54.

reconbot, 2017-10-17:
You need to use election rebuild. Please see the electron section in the
docs.

peiyc, 2017-10-18:
@reconbot I tried it but didn`t work

reconbot, 2017-10-18:
What did you try?

reconbot, 2017-10-18:
In any case if this doesn't work please open an issue with electron-rebuild. We're setup how they've advised.

closed 2017-10-18 by reconbot