serialport/node-serialport #1369: puts raspberry slow

joingood, 2017-10-19:
- __SerialPort Version__: last version
- __NodeJS Version__: 8.0.0
- __Operating System__ and __Hardware Platform__: raspbian

### Summary of Problem
[español]
Estoy utilizando una raspberry pi, entones tengo problemas con la comunicancion serial, el puerto esta abierto pero no resivo nada y no envio nada por serial.

Cuando entro al taskManager, serial port en la parte de resepcion de datos [ port.on("data",callback) ] se come la ran poco a poco, esto hace que la pi se ponga lenta.

matthova, 2017-10-20:
@joingood puedes hacer un ejemplo?
Esta tan lenta cuando usas un parser tan como [esto](https://github.com/matthova/serialport-6-250000-test/blob/master/index.js#L16)?

joingood, 2017-10-20:
El problema era que, al iniciar el nodejs, desde el no se resivia ni se enviaba por serial, yo estoy ulilizando en ttyAMA0 para la conexion serial, entonces al iniciar el node la ran se llenaba poco a poco y el procesador iba muy forzado.

Ya lo solucione, lo que pasaba es que no estaba configurado el ttyAMA0 como principal en el serial del raspbian, estaba ttyS0 como principal.

Lo solucione con este link:
https://spellfoundry.com/2016/05/29/configuring-gpio-serial-port-raspbian-jessie-including-pi-3/

matthova, 2017-10-20:
Bueno, entonces podemos cerrar este issue?
Deberiamos sugerir documentacion adicional para ayudar a otros a evitar este problema?

Cool, can we close this issue?
Is there additional documentation we can add to help other avoid this problem?

joingood, 2017-10-21:
Si, ya se puede cerrar ese issue. 

Y deben de poner esto en la documentacion, ya que seguro que le pasara esto a los nuevos usuarios de esta libreria cuando esten usando el ttyAMA0 en raspbian ya que este tty es el unico que me ha funcionado y en raspbian por defecto el que viene es el ttyS0 (no funciona con esta libreria).

Mas detallado de lo que me paso:

Instale muy bien la libreria en raspberry pi 3 con el OS Raspbian Jessie, en la configuracion del modulo utilizo el /dev/ttyAMA0, cuando ejecute el node no me daba ningun error ni lee ni escribe por serial, entoces el pi al pasar unos minutos, se me frizo, entre al task manager y vi que la ran estaba casi al maximo y el procesador estaba forzado, cuando deteni el node (app), la ran bajo a un 95% y el procesador tambien y vi que era el serial ya que comente todo los otros codigos y solo deje el serial, Luego que note que era el serial busque por intenet y encontre esta solucion en la web que soluciono el problema:

https://spellfoundry.com/2016/05/29/configuring-gpio-serial-port-raspbian-jessie-including-pi-3/

closed 2017-10-21 by joingood