serialport/node-serialport #1370: Unexpected behaviour of serialport when used in a Typescript program

HemantJoshi, 2017-10-22:
I am testing serialport in typescript.

serialport version - 6.0.0
node version - 6.11.3
OS - Linux Ubuntu 14.04, X86_64
Serial HW - Prolific PL2303 used with external loopback

### Summary of Problem
The program is expected to write 32 bytes in one go and stop. The same data is expected to be received. The program gets compiled w/o error. Program also gets executed w/o error.
When I ran the transpiled .js file, the program runs for lot longer (looks like write method is executed more than once).
Note that if I use one of the examples at https://www.npmjs.com/package/serialport in Javascript/node, the program works. 

### Steps and Code to Reproduce the Issue
In a terminal, 
1. tsc to compile the program.
2. Use node to run the js file.

Find the source code below.
My typescript file:

var SerialPort = require ('serialport');

const tx_array1 = new Buffer([1]);
const tx_array2 = new Buffer([0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15,
                             16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31]);

var rx_buffer = new Buffer(32);

class HeadComm {
    path: string;
    baud_rate: {};
    port: any;
    
    
    constructor(_path: string, _baud_rate: {}) {
         this.path = _path;
         this.baud_rate = _baud_rate;
         this.port = new SerialPort(this.path, this.baud_rate);
         
         /*
         SerialPort.list(function (err, ports) {
            ports.forEach(function(port) {
              console.log(port.comName);
              console.log(port.pnpId);
              console.log(port.manufacturer);
            });
        });
        */

    } //constructor
    
    /*
    sendCommand() {
        this.port.write(Buffer.from(tx_array2.buffer));
    }
    */
    
        
}

var head_comm = new HeadComm("/dev/ttyUSB0", {baudRate:9600});
console.log(head_comm.path); console.log(head_comm.baud_rate);

head_comm.port.on('readable', function () {
    
      //console.log(port.read(32));
      rx_buffer = head_comm.port.read(32);
      if (rx_buffer != null) {
          console.log("Received data    =", rx_buffer);
          
      }
});

head_comm.port.write(Buffer.from(tx_array2.buffer));

reconbot, 2017-10-26:
You might want to use the `data` event if you'd like to log all incoming data. I also recommend upgrading to [`serialport@6.0.4`](https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/tag/v6.0.4) we have a windows bug with lower 6x versions.

HemantJoshi, 2017-10-26:
This is what I tried as per your suggestion.
1. Installed serialport@6.0.4
2. Used 'data' event, although 'readable' event is desired.

With 'data' event, however I am seeing the same behaviour as reported earlier.

As I said earlier, following node program works well. It is the Typescript that is causing the trouble.

var SerialPort = require('serialport');
var port = new SerialPort('/dev/ttyUSB0', {
  baudRate: 57600
});

var tx_array1 = new Uint8Array([48, 69, 20, 48, 0x6f, 77, 20, 41, 72, 65, 20, 79, 0x6f, 75]);
var tx_array2 = new Uint8Array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15,
                                16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31]);
var rx_array = new Uint8Array(14);
var rx_buffer = new Buffer(32);

port.on('readable', function () {

  rx_buffer = port.read(32);
  if (rx_buffer != null) {
      if (rx_buffer[0] == 0) {
          console.log("Received data    =", rx_buffer);
      }
  }
});

//raw data
port.write(Buffer.from(tx_array2.buffer));
console.log("Transmitted data =", Buffer.from(tx_array2.buffer));

leojim06, 2017-12-28:
how can I catch all the data from de ```data``` event in just a single var 

For example 

``` javascript
let chunks = [];

// Get all data in chunks array
port.on('data', data => chunks.push(data));
// in the 'end' event return a single string
port.on('end', () => Buffer.concat(chunks).toString());
```

The ```end``` event is not suported... or maybe there is another solution to return the full data sent

>You might want to use the data event if you'd like to log all incoming data. I also recommend upgrading to serialport@6.0.4 we have a windows bug with lower 6x versions.
>
>For example
>
>```port.on('data', data => console.log(data.toString('hex')))```
>The readable event will only fire after read has been called until the buffer is empty, so you have to keep calling read, wait until you get null and then you can wait for another readable event before you start calling read again.

reconbot, 2018-08-29:
you can append the data to a string until you close the port. There is no end because this isn't a request/response stream but one you open and close yourself. Feel free to open a separate issue bout this. 

Closing due to age.

closed 2018-08-29 by reconbot