serialport/node-serialport #1371: Pre-built releases for ARM (Raspberry) highly anticipated

fritzdimmel, 2017-10-22:
- __SerialPort Version__: 6.0.1
- __NodeJS Version__: 8.7.0
- __Operating System__ and __Hardware Platform__: Linux / ARM

- Have you checked the right __version__ of the __api docs?__: Y
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? N
- Are you using __Electron__ and have you checked the _Electron Docs_?: N

### Summary of Problem
It would be awesome, if you could provide pre-built binaries for ARM v6/v7 to use serialport on Raspberry Pi. 
### Steps and Code to Reproduce the Issue
Each install requires complete compile process and takes quite long.

pi@raspberrypi:~/myrepo $ npm install .

> serialport@6.0.1 install /home/pi/myrepo/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install info begin Prebuild-install version 2.3.0
prebuild-install info looking for local prebuild @ prebuilds/serialport-v6.0.1-node-v57-linux-arm.tar.gz
prebuild-install info looking for cached prebuild @ /home/pi/.npm/_prebuilds/https-github.com-EmergingTechnologyAdvisors-node-serialport-releases-download-v6.0.1-serialport-v6.0.1-node-v57-linux-arm.tar.gz
prebuild-install http request GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/v6.0.1/serialport-v6.0.1-node-v57-linux-arm.tar.gz
prebuild-install http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/v6.0.1/serialport-v6.0.1-node-v57-linux-arm.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=8.7.0 runtime=node arch=arm platform=linux)
make: Entering directory '/home/pi/myrepo/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/home/pi/myrepo/node_modules/serialport/build'
added 222 packages in 78.25s

Thank you very much!

reconbot, 2017-10-22:
There currently isn't a way to detect which version of arm node is running
on. (It misreports it and it isn't included in the usual process.arch) so
we can't as much as I wish we could. Hopefully it will be worked out in the
next few node versions.

fritzdimmel, 2017-10-22:
Hi,
thanks for the fast reply. Understood.
Hopefully it will improve in the future!
Thanks,
closing the issue.
Fritz

closed 2017-10-22 by fritzdimmel