serialport/node-serialport #1375: Adding listener to read data causes node to exit

neimanpinchas, 2017-10-22:
- __SerialPort Version__:6.0.3
- __NodeJS Version__:v6.11.2
- __Operating System__ and __Hardware Platform__:win 10 x86_64

- Have you checked the right __version__ of the __api docs?__:?

### Summary of Problem
i am able to write to the serial port
when i am doing a port.on("readable or on("data the script ends immediately, even if the event queaue is not empty

### Steps and Code to Reproduce the Issue
```javascript
var SP= require('../');

var port = new SP("COM2");

port.on('error', function(err) {
  console.log('Error: ', err.message);
})

//when this is uncomented the script keeps running
port.on('data', function () {
  console.log('Data:');
});  

port.on('open', () => {
  console.log(`echo: Port open: COM2`);

    console.log('echo: READY!');

  //port.write("a");

 
});

//code to keep the event loop running
(function wait () {
   if (true) setTimeout(wait, 1000);
})();

reconbot, 2017-10-23:
@kimushu can you take a look? There might be another event loop issue.

kimushu, 2017-10-24:
oh,,, sorry. I made a big mistake about reading. I'll send a pull request soon.

neimanpinchas, 2017-10-24:
For the time being, after googling, i used core node .js for that, it
worked, and i am wondering... (maybe its a new feature of node?, or its not
cross platform?)

var fs = require('fs');
var buffer=""

fs.open("\\\\.\\COM2","r+" ,function (err, fd) {

    if (err) {
        return console.error(err);
    }

    var buffr = new Buffer(1);
    fs.write(fd,"aaa")
function nextread(){
fs.write(fd,"@C"+buffer)
    fs.read(fd, buffr, 0, buffr.length, 0, function (err, bytes) {

        if (err) throw err;

        // Print only read bytes to avoid junk.
        if (bytes > 0) {
            console.log(buffr.slice(0, bytes).toString());
buffer+=buffr.slice(0, bytes).toString()
console.log(buffer)
nextread()
        }

    });
}
nextread()
});

@C clears the display and echos the whole string (for my purpose)

In my raspberry pi i managed with "/dev/ACM0"

Are there any benefits of going C?

PR #1377 (2017-10-24): Fix parameter for ReadThread (windows)
Fix #1375 
This is a degrade by #1367
closed 2017-10-24 by reconbot