serialport/node-serialport #1379: Port open sometimes fails on Windows

kimushu, 2017-10-24:
- __SerialPort Version__: 6.0.3 + #1377 
- __NodeJS Version__: 6.9.5
- __Operating System__ and __Hardware Platform__: Windows 10 Pro x64 + Arduino UNO R3

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? no
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem
Found in #1337.
On some PC, the integration test fails at `2k test`. The failure rate varies depending on PCs.
```
   :
  win32 SerialPort Integration Tests
    static Method
      .list
        √ contains the test port
    constructor
      √ provides an error in callback when trying to open an invalid port
      √ emits an error event when trying to open an invalid port
    opening and closing
      √ can open and close
      √ cannot be opened again after open
      √ cannot be opened while opening
      √ can open and close ports repetitively (60ms)
      √ can be read after closing and opening (3250ms)
      √ errors if closing during a write
    #update
      - Feature "port.update-baudrate" is disabled in "win32. "allows changing the baud rate of an open port"
    #read and #write
      1) 2k test
  :
  1) win32 SerialPort Integration Tests
       #read and #write
         2k test:
     Error: Opening COM15: Access denied
      at Error (native)
```

### Steps and Code to Reproduce the Issue
Run `TEST_PORT=$(./bin/find-arduino.js) npm test`

kimushu, 2017-10-24:
help wanted

My guess:
In the previous test (`errors if closing during a write`), the closing handle during write will be performed _asynchronously_ because [CancelIoEx](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/v6.0.3/src/serialport_win.cpp#L568) API does not guarantee complete of I/O cancelling.
So if there are pending write/read threads and cancelling does not finish before [CloseHandle](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/v6.0.3/src/serialport_win.cpp#L570), the callback of port.close() may be called _before_ actual closing the file handle for COM port.
Because Windows always locks (=opens exclusively) the COM port handle, the following port.open() may fail with "Access denied".
Adding some delays just after `errors if closing during a write` test is very effective.

reconbot, 2017-10-26:
Ok, I hate delays because they're often unreliable but that sounds perfectly reasonable.

closed 2018-05-06 by reconbot