serialport/node-serialport #1380: Can't listen 'data' on windows V6.0.3

antoniobarbosa, 2017-10-25:
- __SerialPort Version__:6.0.3
- __NodeJS Version__:6.11.4
- __Operating System__ and __Hardware Platform__:Windows

- Have you checked the right __version__ of the __api docs?__:yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?:

### Summary of Problem
When I create the listener 
`	port.on('data', function (data) {
		}'...`
the application suddenly closes without messages or errors.
I solve the issue rollbacking to the version 6.0.0
### Steps and Code to Reproduce the Issue

matiasdecarli, 2017-10-26:
I have the very same issue only on Windows. Linux and Mac works as charm

reconbot, 2017-10-26:
Please try [serialport@6.0.4](https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/tag/v6.0.4) Should be fixed.

antoniobarbosa, 2017-10-26:
It's Working fine right now! Thank you

closed 2017-10-26 by antoniobarbosa