serialport/node-serialport #1382: sxsxsxsx

mridah, 2017-10-29:
- __SerialPort Version__: 6.0.4
- __NodeJS Version__:  v8.8.1
- __Operating System__ and __Hardware Platform__: Windows 10 64 bit, 4GB ram

I'm using this library in electron on Windows 8. I've ran `.\node_modules\.bin\electron-rebuild.cmd
`  and even tried adding `"rebuild": "electron-rebuild -f -w serialport"` to the package.json but still can't get it to work.

It gives the following error when I run `npm start`:

> App threw an error during load
Error: A dynamic link library (DLL) initialization routine failed.
\\?\D:\updatedElec\node_modules\serialport\build\Release\serialport.node
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at bindings (D:\updatedElec\node_modules\bindings\bindings.js:81:44)
    at Object.<anonymous> (D:\updatedElec\node_modules\serialport\lib\bindings\win32.js:2:36)

vicatcu, 2017-10-29:
Probably you need python 2.7 sms visual studio community edition installers.

vicatcu, 2017-10-29:
Probably you need python 2.7 and visual studio community edition installed.

mridah, 2017-10-29:
@vicatcu thanks for the fast reply. I already have these installed.

fenixphp, 2017-11-22:
The same problem is one in one. If there is a solution, thanks in advance.

hanfengcan, 2017-11-22:
@fenixphp rebuild in electron env

export(linux) or set(win) this 

```
export npm_config_target=1.2.3
# Electron 的系统架构, 值为 ia32 或者 x64。
export npm_config_arch=x64
export npm_config_target_arch=x64
# 下载 Electron 的 headers。
export npm_config_disturl=https://atom.io/download/electron
# 告诉 node-pre-gyp 是为 Electron 构建。
export npm_config_runtime=electron
# 告诉 node-pre-gyp 从源代码构建模块。
export npm_config_build_from_source=true
```
https://electronjs.org/docs/tutorial/using-native-node-modules#通过-npm-安装

and then use `npm install serialport --save` to install

closed 2018-05-06 by reconbot