serialport/node-serialport #1384: Nan dependency

farfromrefug, 2017-11-01:
Shouldn't ```nan``` only be required in ```devDependencies```? Once serialport is built i don't think you need it.

fivdi, 2017-11-01:
If prebuilt binaries were available for download for all platforms then this might be possible. However, for ARM platforms like the Raspberry Pi and BeagleBone there are no prebuilt binaries available and `serialport` needs to be compiled when it's installed. In such cases `nan` is still needed in `dependencies`.

farfromrefug, 2017-11-01:
@fivdi but it's a dev thing then? Building the module is by definition a dev thing. Thus the nan should only be in dev deps

fivdi, 2017-11-01:
@farfromrefug `nan` isn't only needed during development. It's also needed when serialport is installed by end users on their machines and no prebuilt binaries are available. If `nan` was in devDependencies and somebody did an `npm install serialport` then `nan` would not be installed and `serialport` would not compile. `npm install serialport` doesn't install any devDependencies.

reconbot, 2017-11-01:
That's 💯% correct, keeping it ensures that SerialPort can be used in a much larger set of situations.

closed 2017-11-01 by reconbot

farfromrefug, 2017-11-01:
@fivdi @reconbot If you look at the npm install doc:
https://docs.npmjs.com/cli/install
It says that devDeps are not installed if you use ```--production``` so doing ```npm install serialport``` would install them, right?
I might be wrong. And if i am i am sorry to bug you!

reconbot, 2017-11-01:
No worries, there's another scenario. When you install a package as a dependency of your project, it will only install the the package's production dependencies. The flags you're referencing correspond to your project.

farfromrefug, 2017-11-01:
@reconbot got it!
This is not a big deal with nan as it is not that big. But it is an issue in general when you want to build then publish like you would do with Electron. You end having a lot of deps not actually used at runtime.
Will report on NPM's side.

Thanks!

reconbot, 2017-11-01:
There's not much to report. If you're packaging for electron than you can guarantee the binary will exist and you can remove the install time decencies manually. You'll also want to remove `prebuild-install` as that is responsible for downloading the binaries.

farfromrefug, 2017-11-01:
I am talking about the fact that all the "build" dependecies will be packed too. 
And yes i can remove manually ```prebuild-install``` but in other cases like ```node-pre-gyp``` which have deep dependencies this gets almost impossible.