serialport/node-serialport #1385: Always read the serial port's info

Ju2ender, 2017-11-01:
- __SerialPort Version__: 6.0.0
- __NodeJS Version__: 6.11.4
- __Operating System__: Windows 7 x64 
- __Hardware Platform__: Dini Argeo scales-weighing system

### Summary of Problem

Hi, there

I've a electronic scales. I cat get the weight data when I press the print button on scales. 

But how can I get the data without press the print button? Always listen the data change?

Since I also need to print the weight, I need to press the button twice (print button on scales, and print button on web page).

```JavaScript
var SerialPort = require("serialport");

var port = new SerialPort("COM1", {
    baudRate: 9600,
    dataBits: 8,
    parity: "none",
    stopBits: 1
}, function (error) {
    if (error) {
        console.log(error);
    } else {
        console.log("isOpen:", port.isOpen);
        console.log("path:", port.path);
    }
});

var weight = "";
var currentIndex = 0;

port.on("data", function (data) {
    var buffer = Buffer.from(data, "ascii");

    weight += buffer.toString();
    currentIndex++;

    if (currentIndex == 3)
    {
        weight = weight.trim();

        console.log("Data:", weight);

        currentIndex = 0;
    }
});
```

reconbot, 2017-11-01:
Sounds like it's not sending the data unless you push the button. There might be a setting on the scale to always send the weight. I don't know about your webpage.

fivdi, 2017-11-01:
It may be possible to send requests for information or commands to perform actions to the scales via it's serial port. Find out if there is a datasheet, user guide or some type of specification for the scales. This documentation should indicate what's possible with the serial port.

Ju2ender, 2017-11-02:
Thanks a lot :)
It can receive the READ command.

fivdi, 2017-11-02:
@Ju2ender 
> Thanks a lot :)

You're welcome.

> It can receive the READ command.

That's good news.

It sounds like it's working now so I'll go ahead and close the issue.

closed 2017-11-02 by fivdi