serialport/node-serialport #1386: NODE_VERSION_MODULE error

semonec, 2017-11-01:
- __SerialPort Version__: v6.0.3
- __NodeJS Version__:  v7.1.0
- __Operating System__ and __Hardware Platform__: Mac OSX 10.11 El Capitan

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? yes
- Are you using __Electron__ and have you checked the _Electron Docs_?: not using electron

### Summary of Problem
installing serialport with clang, then running that task, below error apperas

`Here is the error stack:  Error: The module '/Users/sempark/mbed/vscode_extension_mbed/node_modules/serialport/build/Release/serialport.node'
extensionHostProcess.js:28
was compiled against a different Node.js version using
NODE_MODULE_VERSION 51. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
	at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
	at Object.Module._extensions..node (module.js:598:18)
	at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
	at Module.load (module.js:488:32)
	at tryModuleLoad (module.js:447:12)
	at Function.Module._load (module.js:439:3)
	at Function.G.X.o._load (/Users/sempark/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/node/extensionHostProcess.js:28:422043)
	at Module.require (module.js:498:17)
	at require (internal/module.js:20:19)
	at bindings (/Users/sempark/mbed/vscode_extension_mbed/node_modules/bindings/bindings.js:81:44)`

### Steps and Code to Reproduce the Issue
`$CXX=clang npm i serialport`

then run. that's all

fivdi, 2017-11-01:
Are you sure you are not using Electron at runtime as it appears to be the case. The NODE_MODULE_VERSION for Node.js v7.1.0 is 51 (see https://nodejs.org/en/download/releases/). There is node version of Node.js that uses NODE_MODULE_VERSION 54. Electron does however appear to use NODE_MODULE_VERSION 54.

fivdi, 2017-11-21:
I'm going to assume this was an installation/configuration issue and go ahead and close this issue. If it turns out to be an issue with serialport the issue can of course be reopened.

closed 2017-11-21 by fivdi