serialport/node-serialport #1390: Uncaught SyntaxError: serialport.js:146

mishkeTz, 2017-11-03:
SerialPort: 6.0.4
Node: 8.7.0
OS: Win10

Before I instealled serialport I had installed 
```
npm install -g node-gyp
npm install --global --production windows-build-tools
npm install serialport --build-from-source
```

I'm trying to get a list of serial ports on my laptop with nw.js
 
When I compile the code and run it I get the following error in console: 

> Uncaught SyntaxError: C:\Users\nikol\AppData\Local\Temp\nw8832_8531\node_modules\serialport\lib\serialport.js:146
>   FLOWCONTROLS.forEach((control) => {
>                                  ^^
> Unexpected token =>

And this is the code:

```
var SerialPort = require('serialport');
// callback approach
SerialPort.list(function (err, ports) {
  ports.forEach(function(port) {
    console.log(port.comName);
    console.log(port.pnpId);
    console.log(port.manufacturer);
  });
});
```

fivdi, 2017-11-03:
The error message "Unexpected token =>" is related to arrow functions which were introduced with Node 4. It looks like a very old version of node is being used (< Node 4), not Node 8.7.0.

If the following line is added as the first line of the program and the program is then run what is printed to the screen?
```js
console.log("Node " + process.version)
```

fivdi, 2017-11-03:
It shouldn't be necessary to run the following commands.
```
npm install -g node-gyp
npm install --global --production windows-build-tools
npm install serialport --build-from-source
```
Your system may not be setup correctly.

mishkeTz, 2017-11-06:
Oh my God. 
It says Node v1.2.0

fivdi, 2017-11-06:
There are many ways to install the latest version of Node. One way it to download the latest version for your platform at https://nodejs.org/en/ and install it. It may be a good idea to uninstall the version you currently have installed first.

I'll go ahead and close this issue as there is no actionable for serialport.

closed 2017-11-06 by fivdi