serialport/node-serialport #1391: Get return variable from function

flint002, 2017-11-04:
help make this works
```
var result =  sendread("AT+GMR") 
console.log(result) //undefined

    function sendread(input){
    port.write(input+'\r\n')
    port.drain(() => {
     port.once('data',(datas) =>{
         if (datas.toString().match(/"OK"/g)) {
            return 'Not OK'
           }
           return datas.toString() 
     })
     })
     }

```

Thank you

flint002, 2017-11-04:
from https://github.com/node-serialport/node-serialport/issues/1069
now working thanks
```
port.on('open', function() {

      sendread("AT+GMR", function(res) {
        var result =  res
        console.log(result) //652a09gg.Q2406B 1489876 060706 17:19 OK
      })

    })

    function sendread(input, cb) {
      port.write(input + '\r\n')
      port.drain(() => {
        port.once('data', (data) => {
          if (data.toString('utf8')!=="OK") {
            cb("not OK");
          }
          cb(data.toString('utf8'));
        })
      })
    }
```

reconbot, 2017-11-04:
I'd call `port.pause()` after the data event otherwise you'll lose data

flint002, 2017-11-06:
you mean lose data like this ?

```javascript
<script>
  'use strict';
  const SerialPort = require('serialport');
  const parsers = SerialPort.parsers;
  const parser = new parsers.Readline({
    delimiter: '\r\n'
  });
  const port = new SerialPort('COM9', {
    baudRate: 115200,
  });
  port.pipe(parser);

  sendread("AT", function(res) {console.log("command 1 " + res)})
  sendread("AT+GMI", function(res) {console.log("command 2 " + res)})
  sendread("AT+GMR", function(res) {console.log("command 3 " + res)})
  sendread("AT+CSQ", function(res) {console.log("command 4 " + res)})
  sendread("AT+COPS?", function(res) {console.log("command 5 " + res)})

  function Xsendread(input, cb) {
    port.write(input + '\r\n')
    port.drain()
    parser.once('data', (data) => {
      cb(data);
    })
  }
/* Result
serial version 5.0.0
index.html:23 command 1 OK
index.html:24 command 2 OK
index.html:25 command 3 OK
index.html:26 command 4 OK
index.html:27 command 5 OK
*/
  function sendread(input, cb) {
    port.write(input + '\r\n')
    port.drain(() => {
      parser.once('data', (data) => {
        cb(data);
      })
    })
  }
  /* Result
serial version 5.0.0
index.html:23 command 1 OK
index.html:26 command 4  WAVECOM MODEM
index.html:27 command 5  WAVECOM MODEM
index.html:24 command 2  WAVECOM MODEM
index.html:25 command 3  WAVECOM MODEM
*/

  console.log('serial version ' + require('serialport/package').version)
</script>
```
can you help me solve this ? two day i search similar case like this with no luck
i need send a lot command to modem
each function result i wanna store to variable
thank you

SunlightBLR, 2017-11-21:
@flint002 did you fix this?

flint002, 2017-11-21:
not yet
any idea ?

reconbot, 2018-01-15:
Each data event does not have a guarantee to have all available data. The data might be received in multiple events. I would not use the `once` but use `on` or use the `readable` event with the `read()` function. You can find out more in the stream handbook in English and Chinese. https://github.com/substack/stream-handbook#stream-handbook

closed 2018-01-15 by reconbot