serialport/node-serialport #1394: An unhandled error occurred inside electron-build

mridah, 2017-11-05:
- SerialPort Version: ^6.0.4
- NodeJS Version: v8.8.1
- Operating System and Hardware Platform: Windows 10 64 bit, 4GB ram

This is my `package.json`:

      "scripts": {
         "start": "electron .",
        "postinstall": "install-app-deps",
        "dist": "build",
        "install": "electron-rebuild"
      },
    "devDependencies": {
        "electron": "latest",
        "electron-builder": "latest",
        "electron-rebuild": "^1.6.0"
    },

I'm getting this error when I run `npm install` after adding the `electron-rebuild` install hook

> An unhandled error occurred inside electron-rebuild
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
C:\Users\KCS\Desktop\updatedElec\node_modules\serialport\build\serialport.vcxproj(20,3): error MSB4019: The imported project "C:\Microsoft.Cpp.Default.props" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk.
gyp ERR! build error
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\KCS\Desktop\updatedElec\node_modules\node-gyp\lib\build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:125:13)
gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Windows_NT 10.0.14393
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\KCS\\Desktop\\updatedElec\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=1.7.9" "--arch=x64" "--dist-url=https://atom.io/download/electron" "--build-from-source"
gyp ERR! cwd C:\Users\KCS\Desktop\updatedElec\node_modules\serialport
gyp ERR! node -v v8.8.1
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok

> Failed with exit code: 1

> Error: Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
C:\Users\KCS\Desktop\updatedElec\node_modules\serialport\build\serialport.vcxproj(20,3): error MSB4019: The imported project "C:\Microsoft.Cpp.Default.props" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk.
gyp ERR! build error
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\KCS\Desktop\updatedElec\node_modules\node-gyp\lib\build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:125:13)
gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Windows_NT 10.0.14393
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\KCS\\Desktop\\updatedElec\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=1.7.9" "--arch=x64" "--dist-url=https://atom.io/download/electron" "--build-from-source"
gyp ERR! cwd C:\Users\KCS\Desktop\updatedElec\node_modules\serialport
gyp ERR! node -v v8.8.1

How to solve this ???

mridah, 2017-11-08:
Guys @reconbot @voodootikigod ????

reconbot, 2017-11-08:
No idea dude, looks like an electron builder issue.

mridah, 2017-11-13:
@reconbot any idea how can I resolve it ??

reconbot, 2017-11-13:
I'd open an issue with electron builder

TheCbac, 2017-11-16:
@mridah  Do you have the Visual Studio build tools installed? Try:

`npm install --global --production windows-build-tools`

And then building again

phantompepper, 2017-12-02:
Money. This worked for me! Thanks!

hanfengcan, 2017-12-02:
look at here https://github.com/node-serialport/node-serialport/issues/1410

closed 2018-02-05 by reconbot