serialport/node-serialport #1395: No darwin version for 4.0.7 and Node v57

IncognitoCore, 2017-11-06:
### Versions:
`node -v` v8.7.0
`npm -v` 5.4.2
`yarn -v` 1.2.1

### Summary of Problem
There is no darwin nor linux version for serialport 4.0.7 for node v57. This makes npm/yarn fail.
I think this should be fixed since 4.0.7 is still supported and also the latest 4.x version.

### Steps and Code to Reproduce the Issue
`yarn add serialport@4.0.7` or `npm install serialport@4.0.7 --save` on a Mac.

### Output of command:
```
node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.39
node-pre-gyp info using node@8.7.0 | darwin | x64
node-pre-gyp info check checked for "/Users/kevin/Developer/*/node_modules/serialport/build/Release/serialport.node" (not found)
node-pre-gyp http GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-darwin-x64.tar.gz
node-pre-gyp http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-darwin-x64.tar.gz
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-darwin-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@8.7.0 (node-v57 ABI, unknown) (falling back to source compile with node-gyp)
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-darwin-x64.tar.gz
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp clean' (Error: spawn node-gyp ENOENT)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/Users/kevin/Developer/*/node_modules/node-pre-gyp/lib/util/compile.js:77:29)
node-pre-gyp ERR! stack     at emitOne (events.js:115:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:210:7)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
node-pre-gyp ERR! stack     at onErrorNT (internal/child_process.js:374:16)
node-pre-gyp ERR! stack     at _combinedTickCallback (internal/process/next_tick.js:138:11)
node-pre-gyp ERR! stack     at process._tickCallback (internal/process/next_tick.js:180:9)
node-pre-gyp ERR! System Darwin 16.7.0
node-pre-gyp ERR! command "/usr/local/Cellar/node/8.7.0/bin/node" "/Users/kevin/Developer/*/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /Users/kevin/Developer/*/node_modules/serialport
node-pre-gyp ERR! node -v v8.7.0
```

IncognitoCore, 2017-11-06:
Reinstalling node fixed this issue. Sorry for the inconvenience! 😃

closed 2017-11-06 by IncognitoCore

reconbot, 2017-11-06:
No worries, We don't support versions older than 6 anyway. It's a ton of
work to backport the build processes.

---
Francis Gulotta
wizard@roborooter.com

IncognitoCore, 2017-11-06:
Aah I understand. I'll just use the newer version for my own project then! The older version was referenced in some other guys library in his package.json, which I was using for testing :)