serialport/node-serialport #1397: All ARCH="x86" builds on Travis are being built with Node.js 7

Lange, 2017-11-10:
Hi there! :wave:

I was helping the fine folks over at [MadLittleMods/node-usb-detection](https://github.com/MadLittleMods/node-usb-detection) set up `prebuild`, and I've been using the CI configurations of this project as a base.

However, I believe I have discovered an error. It seems that all the `ARCH="x86"` builds are being built on Node.js 7, instead of the desired version. I believe this is because of [`.travis.yml#L43`](https://github.com/node-serialport/node-serialport/blob/master/.travis.yml#L43), which uses the value of `process.version` to check what version of Node to fetch. This will always return whatever the default version of Node.js installed on the container is, rather than the intended value of `TRAVIS_NODE_VERSION`.

I believe that the fix would be to edit that line to pull from the `TRAVIS_NODE_VERSION` environment variable, as opposed to pulling from `process.version`.

Thanks for maintaining this wonderful library and for putting together such comprehensive build scripts, they're super helpful!

reconbot, 2017-11-10:
You're absolutely right we're testing the wrong versions!

Lange, 2017-11-11:
Here's what I did to work around it:
```yaml
before_install:

# download node if testing x86 architecture
- >
  if [[ "$ARCH" == "x86" ]]; then
    EXPANDED_VERSION=$(curl https://semver.io/node/resolve/$TRAVIS_NODE_VERSION);
    echo "EXPANDED_VERSION =" $EXPANDED_VERSION
    BASE_URL=https://nodejs.org/dist/v$EXPANDED_VERSION;
    X86_FILE=node-v$EXPANDED_VERSION-$(node -p "process.platform")-x86;
    echo "X86_FILE =" $X86_FILE;
    wget $BASE_URL/$X86_FILE.tar.gz;
    tar -xf $X86_FILE.tar.gz;
    nvm deactivate;
    export PATH=$X86_FILE/bin:$PATH;
  else
    nvm install $TRAVIS_NODE_VERSION
  fi;