serialport/node-serialport #1400: Commands issued via serialport-term work, but simple script doesn't

simoh92, 2017-11-15:
- __SerialPort Version__: 6.0.4
- __NodeJS Version__: 8.9.1
- __Operating System__ and __Hardware Platform__: Raspbian Lite, rpi3

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
I'm trying to control the following [hardware](https://www.openevse.com/files/OpenEVSE_Plus_v4.pdf), which is connected to the rpi GPIO pins.

When I use the command line tool `serialport-term`, I can successfully control the board according to the [instructions provided by the manufacturer](https://openevse.dozuki.com/Guide/Serial+Communications+with+OpenEVSE/13).
example: when I type `$GV`, it returns `$OK 4.8.0 3.0.1^2E` which is the correct firmware version. Also every other command works fine.

However, when I write a simple script to write any command (e.g. '$FS' in the script below), the hardware doesn't respond and the script doesn't return any error. The same command issued by terminal works perfectly.

I've tried all the different options listed in the documents under `serialPort.write(data, [encoding], [callback]) ⇒ boolean` but without success.

### Steps and Code to Reproduce the Issue
This is the code:

```
var serialport = require('serialport');
var portName = '/dev/ttyS0';
var myPort = new serialport(portName, {
   baudRate: 115200
 });
 
var Readline = serialport.parsers.Readline; // make instance of Readline parser
var parser = new Readline(); // make a new parser to read ASCII lines
myPort.pipe(parser); // pipe the serial stream to the parser
 
myPort.on('open', doStuff);
parser.on('data', readSerialData);
myPort.on('close', showPortClose);
myPort.on('error', showError);

function doStuff() {
  console.log('Port open:', myPort.isOpen, ' - Data rate: ' + myPort.baudRate);
  setTimeout(write, 3000)
}

function write(){
  myPort.write('$FS' , function(error) {
    if (error) {
      return console.log('Error on write: ', error.message);
    }
    console.log('message written');
  });
}

function readSerialData(data) {
  console.log(data);
}

function showPortClose() {
  console.log('port closed.');
}

function showError(error) {
  console.log('Serial port error: ' + error);
}
```
and this is the result of `DEBUG=* node index.js`:

```
> serialport:binding:auto-detect loading LinuxBinding +0ms
>   serialport:main opening path: /dev/ttyS0 +0ms
>   serialport:bindings open +0ms
>   serialport:main _read queueing _read for after open +13ms
>   serialport:poller Creating poller +0ms
>   serialport:main opened path: /dev/ttyS0 +6ms
> Port open: true  - Data rate: 115200
>   serialport:main _read reading +5ms
>   serialport:bindings read +24ms
>   serialport:unixRead Starting read +0ms
>   serialport:unixRead waiting for readable because of code: EAGAIN +2ms
>   serialport:poller Polling for "readable" +11ms
>   serialport:main _write 3 bytes of data +3s
>   serialport:bindings write 3 bytes +3s
>   serialport:unixWrite Starting write 3 bytes offset 0 bytesToWrite 3 +0ms
>   serialport:unixWrite write returned null 3 +4ms
>   serialport:unixWrite wrote 3 bytes +1ms
>   serialport:unixWrite Finished writing 3 bytes +0ms
>   serialport:main binding.write write finished +9ms
> message written

```

fivdi, 2017-11-15:
Maybe a newline character is needed to terminate messages sent to the device.

Should the following:
```js
myPort.write('$FS' , function(error) {...
```
perhaps be this:
```js
myPort.write('$FS\n' , function(error) {...
```
or this:
```js
myPort.write('$FS\r' , function(error) {...
```
or maybe this:
```js
myPort.write('$FS\r\n' , function(error) {...
```

simoh92, 2017-11-15:
I've tried all of them but still doesn't work...

simoh92, 2017-11-20:
As I've written above, now I can successfully send commands to the board which responds correctly. 

However, I still haven't been able to read or get any kind of data or response from the device. I know that the commands that I send work, only because they trigger the physical activation of a relay.

When I send a command through terminal (e.g. `$GS`), I receive as response `$OK 3 13^11`, but when I do the same with the script above, I don't receive anything.

I've tried the code above and many other different combinations, but both the 'data' event or port.read() don't return anything. If I use a callback  like:
```
myPort.write('$GS\r'), function(error, res) {
      if (error) {
        return console.log('Error on write: ', error.message);
      }
      console.log('message written', res);
    });,
```
 it always returns undefined. How can I print data from the device? What's wrong with my code?

closed 2017-11-21 by simoh92