serialport/node-serialport #1401: Misses data

SunlightBLR, 2017-11-16:
- __NodeJS Version__: v6.9.5
- __Operating System__ and __Hardware Platform__: Windows 10 64bit

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? no
- Are you using __Electron__ and have you checked the _Electron Docs_?: no

### Summary of Problem

Misses data in port.on('data', ...)

### Steps and Code to Reproduce the Issue
`const SerialPort = require('serialport')

const port = new SerialPort('COM3', {
  baudRate: 57600
})

port.on('open', () => console.log('Port open'))

port.on('data', (data) => {
    console.log(data.toString('utf8'))
})

port.write('AT\n');
`

Code event on('data') Returns only the sent command (port.write('AT')). There is no response from the device. 
Through other programs for testing, the device responds as needed.

**Example**
Sended:
AT
Need to get:
AT
OK

**Using SerialPort**
Sended:
AT
Received:
AT

(without 'OK')

fivdi, 2017-11-16:
Does it work if the following
```js
port.write('AT\n');
```
is replaces with
```js
port.write('AT\r');
```

reconbot, 2017-11-16:
I bet you get them on different data events.

SunlightBLR, 2017-11-16:
@fivdi @reconbot  thank you very much!
Thanks for the unrealistically fast response

closed 2017-11-16 by SunlightBLR