serialport/node-serialport #1404: TypeError: Invalid non-string/buffer chunk

scorpioRED, 2017-11-23:
- __SerialPort Version__: 6.0.4
- __NodeJS Version__: 8.5.0
- __Operating System__ and __Hardware Platform__: Linux / MAC OS

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
Hi! I have try to send command to RFID Reader (StrongLink SL500)
_Bus 006 Device 011: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light_
**/dev/ttyUSB0**

Early I have use chrome extension with _chrome.serial_ and code bellow worked fine.
Now I try to use it on node with serialport module, but I got an error:

```
(node:8646) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): TypeError: Invalid non-string/buffer chunk
(node:8646) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

### Steps and Code to Reproduce the Issue

```
var SerialPort = require('serialport')

var serialPort = new SerialPort('/dev/ttyUSB0', {
    autoOpen: false,
    baudRate: 9600,
    dataBits: 8,
    stopBits: 1,
    parser : SerialPort.parsers.raw
})

var RF_BEEP___________ = [0x06, 0x06, 0x01, 0x10];
var RF_INIT_COM_______ = [0x06, 0x01, 0x01, 0x03];
var RF_GET_MODEL______ = [0x05, 0x04, 0x01];
var GREEN_LED_ = [0x06, 0x07, 0x01, 0x02];

function rfCommandCreate (cmdCode) {
    var byteCommand = [];
    byteCommand[0] = 0xAA; // Command head;
    byteCommand[1] = 0xBB; // Command head
    byteCommand[2] = cmdCode[0]; // Length
    byteCommand[3] = 0x00; // Length
    byteCommand[4] = 0x00; // Deveice ID
    byteCommand[5] = 0x00; // Deveice ID
    byteCommand[6] = cmdCode[1];
    byteCommand[7] = cmdCode[2];
    if (cmdCode.length > 2) {
        for (var i = 3; i < cmdCode.length; i++) {
            byteCommand[byteCommand.length] = cmdCode[i];
        }
    }
    var xor = byteCommand[4] ^ byteCommand[5];
    for (var i = 4; i < byteCommand.length; i++) {
        xor = xor ^ byteCommand[i];
        if (byteCommand[i] == 0xAA) {
            byteCommand.splice(i + 1, 0, 0x00);
            i++;
        }
    }
    byteCommand[byteCommand.length] = xor;
    return byteCommand;
};

function array2bytecode (arr) {
    console.log("runn arr to Byte")
    var bytes = new Uint8Array(arr.length);
    for (var i = 0; i < arr.length; ++i) {
        bytes[i] = arr[i];
    }
    return bytes.buffer;
};

serialPort.open(function (err) {
    if (err) {
        console.log('Error opening port: ', err.message);
        return
    }
    console.log('port opened without error')

   serialPort.write(array2bytecode((rfCommandCreate(RF_INIT_COM_______))));
   serialPort.write(array2bytecode((rfCommandCreate(RF_BEEP___________ ))));
   serialPort.write(array2bytecode((rfCommandCreate(RF_GET_MODEL______))));

});

//Switches the port into "flowing mode"
serialPort.on('data', function (data) {
    console.log('Data:', data);
});
```

fivdi, 2017-11-23:
The `write` method expects a string or [Buffer](https://nodejs.org/dist/latest-v8.x/docs/api/buffer.html) but an `ArrayBuffer` is passed. See `Uint8Array.prototype.buffer` at [Uint8Array](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Uint8Array).

fivdi, 2017-11-30:
I'm going to assume this isn't a serialport issue and go ahead and close the issue. If it turns out to be a serialport issue the issue can od course be reopened.

closed 2017-11-30 by fivdi