serialport/node-serialport #1405: npm install through nodejs referencing missing gz files for the serialport portion

annegast, 2017-11-23:
- __SerialPort Version__: won't install because the file doesn't exist on github
- __NodeJS Version__: 8.9.1
- __Operating System__ and __Hardware Platform__: Windows 10, HP Pavilion DV6

- Have you checked the right __version__ of the __api docs?__:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?: not using Electron

### Summary of Problem
https://github.com/node-serialport/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-win32-ia32.tar.gz is producing a 404 not found error

### Steps and Code to Reproduce the Issue
npm init --yes
npm install johnny-five --save

As it is installing, it calls the URL referenced above to install the serial-port portion and fails. This means the entire installation fails. This also happens with version 9.2.0, but obviously with a different gz file for that version.

fivdi, 2017-11-23:
serialport is a native module containing C++ code that needs to be compiled. To save users the hassle of having to setup their systems to compile serialport when it's installed each version of serialport has prebuilt binaries for several platforms and several versions of Node.js.

`npm install johnny-five --save` will install the latest version of Johnny-Five which is v0.11.7.  Johnny-Five v0.11.7 [depends on serialport v4.0.7](https://github.com/rwaldron/johnny-five/blob/v0.11.7/package.json#L276).

The prebuilt binaries available for serialport v4.0.7 can be seen [here](https://github.com/node-serialport/node-serialport/releases/tag/4.0.7). Unfortunately serialport v4.0.7 doesn't provide prebuilt binaries for Node.js 8.x on 32bit Windows 10 systems as the file `serialport-v4.0.7-node-v57-win32-ia32.tar.gz` doesn't exist.

You have a few options here:

1. Setup your system so that it can compile serialport locally when `npm install johnny-five --save` is run.
2. Downgrade from Node.js v8.9.1 to a version of Node for which serialport v4.0.7 has prebuilt binaries for 32bit Windows Systems. For example Node.js 4 or 6.

reconbot, 2017-11-23:
I've got a hand in firmatta which is what J5 uses under the hood. I had a
branch to upgrade it but it requires some test refactoring and I got
stalled out. I fully support the endevor if someone wants to give it a try.

fivdi, 2017-11-30:
@reconbot can this issue be closed?

reconbot, 2017-11-30:
should be fixed in firmata, we have no easy way to make the newer binary

closed 2017-11-30 by reconbot