serialport/node-serialport #1409: ReadIOCompletion error closes port after flush on Windows

monkbroc, 2017-11-29:
- __SerialPort Version__: 6.0.4
- __NodeJS Version__: v8.7.0
- __Operating System__ and __Hardware Platform__: Windows 10 64-bit

### Summary of Problem

Calling `serialPort.flush` on Windows causes the `Reading from COM port (ReadIOCompletion): Operation aborted` if the port is currently listening which leads to `_disconnect` being called and the port being closed.

I ran into this issue upgrading [the Particle CLI to serialport 6](https://github.com/spark/particle-cli/issues/390).

### Steps and Code to Reproduce the Issue

```
/* Flash this program to a Particle or Arduino device

void setup() {
	Serial.begin(9600);
}

void loop() {
	if (Serial.available()) {
		char c = Serial.read();
		Serial.printlnf("Char %d", c);
	}
}

Install serialport and run this script on Windows
npm install serialport
node index.js

Expected output:
Char 65

write done

Char 65
write done

Actual output:
Char 65

write done
write done
closing
{ Error: Reading from COM port (ReadIOCompletion): Operation aborted disconnected: true }

*/

const SerialPort = require('serialport');

// Replace with the serial port of your device
const port = 'COM16';
const serialPort = new SerialPort(port, {
	baudRate: 9600,
	autoOpen: false
});
serialPort.on('data', data => {
	console.log(data.toString());
});
serialPort.on('close', err => {
	console.log('closing');
	console.error(err);
	process.exit(0);
});
function sendCommand() {
	serialPort.flush(() => {
		serialPort.write('A');
		serialPort.drain(() => {
			console.log('write done');
		});
	});
}

serialPort.open(err => {
	if (err) {
		console.error(err);
		process.exit(1);
	}
	sendCommand();
	setTimeout(sendCommand, 1000);
});
```

mumairofficial, 2018-01-29:
I have same issue... upon `flush ` it close the connection and if I don not flush then second reponse twisted and merged into first response and make invalid response.

@reconbot This is friendly question; Is there any progress on this issue :)

SirUppyPancakes, 2018-03-26:
Also seeing this issue.

Currently, it is important (though not completely a blocking issue) that I flush the buffers before each write during a real-time writeLine-readLine loop where each, last command may fail to respond and could corrupt the next command, and clearing the buffers before each write greatly reduces the chances of corrupted data.

I'll be commenting out my flush() call for now and keeping an eye on the roadmap for the bug-fix, hopefully it will be soon :)

maxwellhadley, 2019-01-23:
Any update on progress fixing this issue?

reconbot, 2019-01-24:
It needs a champion.

PR #1817 (2019-05-02): Removed PURGE_RXABORT flag on flush for Windows
The PURGE_RXABORT flag causes the ReadFileEx to be cancelled, which somehow causes the port to become disconnected (didn't dive that deep into the intricacies of the issue).

After creating this fix, we found issue #1013, which suggests the same resolution; after reviewing PR #1223, I'm not sure that #1013 was ever resolved correctly. 

This resolves #1409.
closed 2019-05-02 by reconbot