serialport/node-serialport #1410: serialport with vscode extensions

ubmtho, 2017-11-29:
serialport 6.0.4
node 6.11.3
Windows 7 professional

### Summary of Problem
I really boils down to a simple question: Should serialport work in an Visual Studio Extension?

I can use serialport with Visual Studio Code, it works really well so thanks for that! The problem occurs when I try do the same from an vscode extension.

var SerialPort = require('serialport');

gives:

Activating extension `a.myext` failed: Could not locate the bindings file. Tried:
extensionHostProcess.js:4
 → c:\Users\a\script\node_modules\serialport\build\serialport.node
 → c:\Users\a\script\node_modules\serialport\build\Debug\serialport.node
 → c:\Users\a\script\node_modules\serialport\build\Release\serialport.node
 → c:\Users\a\script\node_modules\serialport\out\Debug\serialport.node
 → c:\Users\a\script\node_modules\serialport\Debug\serialport.node
 → c:\Users\a\script\node_modules\serialport\out\Release\serialport.node
 → c:\Users\a\script\node_modules\serialport\Release\serialport.node

The path (c:\Users\a\script\node_modules\serialport\Release\serialport.node) do exist. I take it from googling the similar problems might be solved by rebuilding from source, but I have no luck with that. Should I proceed, as the problem is me, or drop vscode and head for something like Atom?

### Steps and Code to Reproduce the Issue

hanfengcan, 2017-11-29:
i use node-serialport in vscode extensions.

[image]

`import * as serialport from 'serialport'`

or you need rebuild node-serialport 

run this .bat 

```
:: Electron 的版本。
set npm_config_target=1.7.9
:: Electron 的系统架构, 值为 ia32 或者 x64。
set npm_config_arch=x64
set npm_config_target_arch=x64
:: 下载 Electron 的 headers。
set npm_config_disturl=https://atom.io/download/electron
:: 告诉 node-pre-gyp 是为 Electron 构建。
set npm_config_runtime=electron
:: 告诉 node-pre-gyp 从源代码构建模块。
set npm_config_build_from_source=true

set PYTHON=D:\Python27
set path=C:\Windows\Microsoft.NET\Framework64\v4.0.30319;%path%

npm install serialport --save
```

ubmtho, 2017-11-29:
That was it! Awesome and thanks a lot for your help!

closed 2017-11-29 by ubmtho