serialport/node-serialport #1411: Serialport output is weirdly wrong.

mrlkn, 2017-11-30:
- __SerialPort Version__:4.0.0
- __NodeJS Version__:6.x.x
- __Operating System__ and __Hardware Platform__: Linux 

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
- Are you using __Electron__ and have you checked the _Electron Docs_?:

### Summary of Problem
```
[var SerialPort = require("serialport");
var serialport = new SerialPort("/dev/ttyACM0",{baudRate:9600});
    
serialport.on('open', function(){
  console.log('Serial Port Opend');
  serialport.on('data', function(data){
      console.log(data[0]);
  });
});
]
```
### Steps and Code to Reproduce the Issue
When i execute this it does not giving my actual serial port data also it brakes the serial port data, when i add something after `{baudRate:9600}` it gives error that ` throw new TypeError('`openImmediately` is now called `autoOpen` and is a property of options');`

fivdi, 2017-11-30:
> When i execute this it does not giving my actual serial port data also it brakes the serial port data

 * What is the expected output of the program and what is the actual output of the program?
 * Is 9600 the correct `baudRate`?
 * Are other options perhaps missing?
 * When the `data` event handler in the example code posted above is called only one byte of the data received is printed to the screen. Why isn't all the data received printed to the screen?

>  when i add something after {baudRate:9600} it gives error that throw new TypeError('openImmediately is now called autoOpen and is a property of options')

In serialport v3 the constructor looked like [this](https://github.com/node-serialport/node-serialport/blob/3.1.2/lib/serialport.js#L86). In serialport v4 it looked like [this](https://github.com/node-serialport/node-serialport/blob/4.0.0/lib/serialport.js#L77-L80).

 * serialport v3 accepted a 3rd parameter called `openImmediately` that was a boolean. This is no longer the case in v4.
 * Are you reading the correct documentaton for v4? It's [here](https://github.com/node-serialport/node-serialport/tree/4.0.0).

fivdi, 2017-12-09:
I'm going assume that all is ok with serialport here and go ahead and close this issue. If it does turn out to be a problem with serialport the issue can of course be reopened.

closed 2017-12-09 by fivdi

mrlkn, 2017-12-19:
Hello aagain and sorry for very late answer i had another issues now i return, i am new at nodejs and arduino. I have an Arduino setup and couple sensors on it. I am reading temperature and humidity values with Arduino. My serial monitor output like this : 

    Humiditiy (%): 44.00
    Temperature (Celcius): 26.00
    Temperature (Kelvin): 299.00
    Temperature (Fahrenheit): 58.00
    Gas Value: 341
    
    Humiditiy (%): 44.00
    Temperature (Celcius): 26.00
    Temperature (Kelvin): 299.00
    Temperature (Fahrenheit): 58.00
    Gas Value: 341
 

**I want three things.**

 1. Using NodeJS and pulling Serial Monitor outputs.
 2. Storing values with MongoDB
 3. Sending values to the website that i created.

I did try to pull the values from serial monitor with this NodeJS file and put the output to the console.

    // Setup basic express server
    var express = require('express');
    var app = express();
    var server = require('http').createServer(app);
    var io = require('socket.io')(server);
    
    
    // Routing
    var SerialPort = require("serialport").SerialPort
    var serialPort = new SerialPort("/dev/ttyACM0", {
        baudrate:115200
    }, false); // this is the openImmediately flag [default is true]
    
    serialPort.open(function () {
     serialPort.on('data', function(data) {
            console.log('Recieving data' + data);
           
        });
        
    });

But the output in terminal is weirdly broke yet this was the best output i get somehow.

     Recieving dataty (%)
    Recieving data: 44.00
    Temperature (Celcius): 26.00
    Temperature 
    Recieving data(Kelvin): 299.00
    Temperature (Fahrenheit): 58.00
    Recieving data
    
    Recieving dataz
    Recieving dataas Value: 380

And now i am working on how to use mongo with serialport any help will be appreciated.

fivdi, 2017-12-19:
The way the program posted above is implemented, serialport will fire `'data'` events as often and as quickly as it can. It will not wait for a complete line of text to be received before firing the `'data'` event which is what I think you are expecting.

The statement being used to output information to the screen is:
```js
console.log('Recieving data' + data);
```
This `console.log` combined with the fact that the program is implemented to result in `'data'` events being fired as often and as quickly as possible result in the following output:
```
 Recieving dataty (%)
Recieving data: 44.00
Temperature (Celcius): 26.00
Temperature 
Recieving data(Kelvin): 299.00
Temperature (Fahrenheit): 58.00
Recieving data

Recieving dataz
Recieving dataas Value: 380
```
If the text "Receive data" and the extra new lines are removed from the output it will look like this (which is better but not fully correct):
```
 ty (%): 44.00
Temperature (Celcius): 26.00
Temperature (Kelvin): 299.00
Temperature (Fahrenheit): 58.00
zas Value: 380
```
There are two errors in the output here. The ` ty` at the start should be `tiy` (the last three letters of `Humiditiy` (which is probably spelt incorrectly and should be Humidity)) and `zas`  which should be `Gas`. I don't know why these errors are occurring and their are many potential reasons. If the Arduino sketch is outputting information as quickly as it can at 115200 baud then the Node.js program is going to have difficulties when attempting to start reading. Perhaps this is the problem. The Node.js program and Arduino should probably coordinate their communication.

If you would like to process lines of text serialport has a concept called parsers. There's a serialport v4.0.0 parser example [here](https://github.com/node-serialport/node-serialport/blob/4.0.0/examples/readdata.js) that should help. An example for serialport v6.0.4 can be found [here](https://github.com/node-serialport/node-serialport/blob/v6.0.4/examples/readline.js). There's also related information in the version specific README files.

I'm not familiar enough with MongoDB to provide any information of use but a google search for tutorials should here here.

mrlkn, 2017-12-19:
Yes thank you parsers are exactly what i needed! My teacher gave me example, which is [https://github.com/celalceken/WaspmoteProject](url) he is using the parsers too. I am gonna give it a try as soon as i can.

jahazieldom, 2018-04-17:
@mrlkn how do you solve this problem? i have similar problem.

mrlkn, 2018-04-18:
@jahazieldom I made my Arduino output like 44,26,299 and parsed them into a Array.