serialport/node-serialport #1412: Serial Port not a constructor Error

waghanin, 2017-12-01:
don't know how to check serialport version
node -v is 8.9.1
OSX Yosemite 10.10.5

I am trying to run "node hello.mag.45.js" and got the error "cannot find module serialport". then I copied the node_modules folder with serialport in it from somewhere else on my computer to the same directory as my js file. Now I get the error that "SerialPort is not a constructor"

fivdi, 2017-12-01:
> don't know how to check serialport version

Calling `npm list serialport` from the command line in the folder containing `hello.mag.45.js` should display the version of serialport being used.

> then I copied the node_modules folder with serialport in it from somewhere else on my computer to the same directory as my js file.

This may not be the best of ideas. Why not start from scratch in a new empty folder and call `npm install serialport` from the command line in that folder to install serialport correctly?

> I am trying to run "node hello.mag.45.js"

Is [this](http://academy.cba.mit.edu/classes/input_devices/mag/hello.mag.45.js) hello.mag.45.js? If so, it looks like it was written a long time ago with a very old version of serialport. To create a serialport object it uses the following code:
```js
var SerialPort = require("serialport").SerialPort
var sp = new SerialPort(serial_port,{baudrate:baud})
```
If the latest version of serialport is used the same code would look like this:
```js
var SerialPort = require('serialport');
var sp = new SerialPort(serial_port,{baudRate:baud});
```
It's highly likely that additional code in hello.mag.45.js would also have to be modified.

fivdi, 2017-12-09:
I'm going assume that all is ok with serialport here and go ahead and close this issue. If it does turn out to be a problem with serialport the issue can of course be reopened.

closed 2017-12-09 by fivdi