serialport/node-serialport #1413: SerialPort Parity Issue on Pi

PiyushRamuka, 2017-12-02:
- __SerialPort Version__: 6.0.4
- __NodeJS Version__: 7.9.0
- __Operating System__ and __Hardware Platform__: Rasp Pi 3

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
I am having similar issue with PI3 Serial Port (ttyS0). The code works without any issues with none parity. It also works with "Even" parity on USB serial port. However, the communication fails with even parity. I don't have any option but to use even parity as the STM32F requires even parity for bootloader. 

I have modified the serialport.cpp as recommend by @voodootikigod  and @sguilly but no luck. Do I need to rebuild after changing the .cpp file? While using "npm rebuild serialport" i get following error - 

error: expected type-specifier before ‘NanUtf8String’
     NanUtf8String *str = new NanUtf8String(v8str);

Here are the plots
[image]

Please let me know how can i resolve this issue. Thanks in advance!

PiyushRamuka, 2017-12-04:
Hello -
We were able to resolve this issue. It was not a software limitation but Pi 3B hardware limitation. Pi3 has 2 UARTs and the primary UART which supports parity is reserved for bluetooth. Hence the code was not working.
Disabling the bluetooth resolved this issue. You can read more on this here - 
[https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=198999&p=1241960#p1241960]
Thank you!

closed 2017-12-04 by PiyushRamuka

reconbot, 2017-12-04:
That's awesome, thank you for figuring this out.