serialport/node-serialport #1415: No data to read on Linux but working on Windows

CapMousse, 2017-12-07:
- __SerialPort Version__: 6.0.4 (both systems)
- __NodeJS Version__: 8.9.2
- __Operating System__ and __Hardware Platform__: *Linux* Elementary OS  - Kernel 4.12 | Windows 10 
- __Targeted hardware__ : 3D printer (MP Select Mini) using GCode.

### Summary of Problem

When trying to communicate with the 3D printer on Windows, everything works fine and I receive all the data from the printer as expected.
When trying the exact same code on Linux, using the same Node and Serialport version, no data is captured and a `EAGAIN` error is triggered.

### Steps and Code to Reproduce the Issue

Run this code on Windows then Linux : 
```js 
const linuxPort = `/dev/ttyACM0`
const windowsPort = `COM3`
const SerialPort = require('serialport')
const parser = new SerialPort.parsers.Readline

const connection = new SerialPort(linuxPort, {
    baudRate: 115200
})

connection.pipe(parser)

parser.on('data', (data) => {
    console.log(data)
})

connection.on('open', _ => {
    connection.write(`M105;\r\n`);
})
```

### Expected result

The printer should send a string of data like : `N0 P15 B15 T:21.0 /0.0 B:17.7 /0.0 T0:21.0 /0.0 @:0 b@:0` representing the temperature of the extruder, the bed and the targeted temperatures but :
- On windows, the data returned is ok
- On linux, nothing append

Apollon77, 2017-12-31:
I have a comparable problem and for me node-serialport 4 is working on linux and the same code with serialport 6 is NOT working.
Can you try on the linux system with serialport 4.0.7?!

Apollon77, 2018-01-02:
@reconbot is anything known with Linux Incompatibilities? I have several projects that uses serialport but I can not upgrade it because then Linux will be broken (at least ARM based systems where I tried it)

fivdi, 2018-01-15:
This issue can be reproduced on Linux.

If the following  echo sketch is runing on an Arduino UNO then I would expect the program from the original post to print `M105;\r\n` to the Linux console when run. However, it doesn't print anything.

```cpp
void setup() {                
  Serial.begin(115200);
}

void loop() {
  if (Serial.available()) {
    Serial.write(Serial.read());
  }
}
```

The following variant of the program does however work:
```js
const linuxPort = `/dev/ttyACM0`
const windowsPort = `COM3`
const SerialPort = require('serialport')
const parser = new SerialPort.parsers.Readline
var count = 0

const connection = new SerialPort(linuxPort, {
    baudRate: 115200
})

connection.pipe(parser)

parser.on('error', (err) => {
    console.log('Error: ', err.message);
})

parser.on('data', (data) => {
    console.log(data)
})

connection.on('open', _ => {
    setTimeout(_ => {
        setInterval(_ => {
            connection.write(count + ` M105;\r\n`);
            count += 1
        }, 100);
    }, 2000);
})
```

Note that this variant of the program waits for 2 seconds before writing data to the port. serialport appears to ignore data written to a freshly opened port for a few seconds before it starts to function correctly.

@CapMousse @Apollon77 can you please modify your program to prevent it from writing to the port for the first few seconds after the `open` event is fired and let us know if it works.

fivdi, 2018-01-15:
baaaaah... I forgot that an Arduino UNO does an automatic reset when it gets a serial connection. This explains why it's necessary to wait for a few seconds before things start to work correctly. It also means that everything appears to be ok with serialport on Linux. Writing and reading data functions correctly.

@CapMousse @Apollon77 Some crucial piece of information that is needed to reproduce the issue is missing from the summary of the problem. What could this information be?

reconbot, 2018-01-15:
The arduino will ignore data while it boots up. It can send data when it is ready, you can use the `ReadyParser` to wait for this data.

Apollon77, 2018-01-15:
But if it is really this (then it would had been an identical problem also with Serialport 4 for you ?!), then my problem is different ... I will still test to wait 2s after opening the port and report. I do communication with an "Energy SMart meter" using an USB IR head. So these problems do not exist there.

reconbot, 2018-01-15:
@Apollon77 I don't know about your setup, it's probably different. I'm still trying to figure out #1399 but maybe you have a related problem to that? I'd open your own issue.

Apollon77, 2018-01-15:
I have opened #1415 :-) 2s wait time is not working for me any I'm on linux and not windows.
see also my last comments on #1415

CapMousse, 2018-01-16:
Sorry for the (very) late reply. The 2 seconds wait work great for my case ! :+1:

fivdi, 2018-01-16:
@CapMousse excellent, thanks for the feedback.

closed 2018-01-16 by fivdi