serialport/node-serialport #1419: Bad file descriptor error when write some byte

mahdi-gh1373, 2017-12-12:
- __SerialPort Version__: latest
- __NodeJS Version__: v 8.8.1
- __Operating System__ and __Hardware Platform__: linux

- Have you checked the right __version__ of the __api docs?__:
- Are you using __Electron__ and have you checked the _Electron Docs_?: yes

### Summary of Problem
I want to write some bytes via AT Command every two min, the file is image and is jpg and it's about 70KB.But after 2 or 3 series the port is closed and I have a bad file decriptor error.
What should I do do now?
### Steps and Code to Reproduce the Issue
```
				loop = setInterval(function(){
				//console.log("picture");
					var picname = "" + LastData.Device_Serial_Number + "_" + parseInt(LastData.Driver_ID, 16) + "_" + LastData.Unix_Time_Stamp;
					console.log(picname);					

                                 Webcam.capture("/root/mytest/app/client-pictures/"+picname, function(err, imgpath){
					
						console.log("new photo")
					
						fs.readFile('/root/mytest/app/clientpictures/'+picname+'.jpg',function(err,data) {
							if (err){
								//document.getElementById("fs").innerHTML = err;
								console.log(err)
							}																		
							// Encode to base64
							var encodedImage = new Buffer(data, 'binary');
							//console.log(encodedImage)
							console.log("inside1")
							var command = 'AT+QFUPL="RAM:'+picname+'.jpg",'+data.length+',30\r';

							mahdiport.write(command, function(err) {
	 							if (err) {
	 								return console.log('Error on write1: ', err.message);
									//document.getElementById("imgpath").innerHTML = "command : "+err;
	 							}
								//document.getElementById("imgpath").innerHTML = "command success";
	 							console.log("inside2")
	 							mahdiport.write(encodedImage, function(err) {
	 								if (err) {
	 									return console.log('Error on write2: ', err.message);
										//document.getElementById("imgpath").innerHTML = "encodedImage : "+err;
	 								}
	 								console.log("inside3")
									//document.getElementById("imgpath").innerHTML = "encodedImage success";
								})
	 						});			
						});		
						
						
					});
				},looptime*90000)
```

mahdi-gh1373, 2017-12-19:
nobody doesn't have any idea about this problem?

reconbot, 2017-12-19:
When the port closes, you should get a close event with an error object. Take a look at that.

mahdi-gh1373, 2017-12-20:
This is an error which I see in my app:
 close : Error: bad file descriptor

closed 2018-11-22 by stale[bot]