serialport/node-serialport #1420: Changing the baud rate of an opened port

Meterius, 2017-12-14:
How can I change the baud rate from a serial port that has been already opened with another baud rate. For example, my usb device starts with the baud rate 9600, but when in use I want the usb device to change its serial communication baud rate to 115200. I tried to do it with port.set({ baudRate: 115200 });, but it did not work and when I logged the error callback, it just said null. What am I doing wrong? (I wrote the Arduino firmware for the usb device myself, so I know that on his part it is changing its baud rate)

reconbot, 2017-12-14:
What operating system and SerialPort version? The correct command is update

Meterius, 2017-12-14:
@reconbot Thank you for the quick reply. The update command is working perfectly.

closed 2017-12-14 by Meterius