serialport/node-serialport #1421: PR: Add: Dynamic Length Parser

frank-dspeed, 2017-12-15:
Will add a Dynamic Length Parser that parses byte buffers based on configured position else use first
it will read the data until length reached and emit as also preserve the bytes after the sequence to start a new seqence

frank-dspeed, 2017-12-15:
## TODO: 
- define pos on init
- better documentation

## Example
```js
'use strict';
const Transform = require('stream').Transform;
const Buffer = require('safe-buffer').Buffer;
/**
 * Parses the CCTalk protocol
 * @extends Transform
 * @example

const SerialPort = require('serialport');
const DynamicByteLength = SerialPort.parsers.DynamicByteLength;
const port = new SerialPort('/dev/ttyUSB0');
const parser = port.pipe(new DynamicByteLength());
parser.on('data', console.log);

 */
class DynamicByteLengthParser extends Transform {
  constructor() {
    super();
    this.array = [];
    this.cursor = 0;
  }
  _transform(buffer, _, cb) {
    this.cursor += buffer.length;
    // TODO: Better Faster es7 no supported by node 4
    // ES7 allows directly push [...buffer]
    // this.array = this.array.concat(Array.from(buffer)); //Slower ?!?
    Array.from(buffer)
      .map((byte) => this.array.push(byte));
    while (this.cursor > 1 && this.cursor >= this.array[pos]) {
      // full frame accumulated
      // copy command from the array
      const FullMsgLength = this.array[pos];

      const frame = Buffer.from(this.array.slice(0, FullMsgLength));
      // Preserve Extra Data
      this.array = this.array.slice(frame.length, this.array.length);
      this.cursor -= FullMsgLength;
      this.push(frame);
    }
    cb();
  }
};

module.exports = DynamicByteLengthParser;
```

reconbot, 2018-02-05:
This is a neat example. Since it's not an issue, I'm going to close it for now.

closed 2018-02-05 by reconbot