serialport/node-serialport #1424: Example with SerialPort.parsers.Ready not working

saewave, 2017-12-16:
- __SerialPort Version__: 6.0.4
- __NodeJS Version__: v8.9.1
- __Operating System__ and __Hardware Platform__: Win7

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
Example with SerialPort.parsers.Ready not working

### Steps and Code to Reproduce the Issue
Use your example to reproduce the issue:

> var SerialPort = require('serialport');
> var Ready = SerialPort.parsers.Ready;
> var port = new SerialPort('COM3');
> var parser = port.pipe(Ready({data: 'READY'}));
> parser.on('data', console.log); // all data after READY is received

Error:
> var parser = port.pipe(Ready({data: 'READY'}));
>                        ^
> TypeError: Class constructor ReadyParser cannot be invoked without 'new'
>     at Object.<anonymous> (E:\Projects\STM32\RemoteSwitch\FWUpdater\testReady.js:4:24)

Ok, adding **new**:
> var parser = port.pipe(new Ready({data: 'READY'}));

After run:
> ...FWUpdater\node_modules\serialport\lib\parsers\ready.js:9
>       throw new TypeError('"delimiter" is not a bufferable object');
>       ^
> 
> TypeError: "delimiter" is not a bufferable object
>     at new ReadyParser (...FWUpdater\node_modules\serialport\lib\parsers\ready.js:9:13)
>     at Object.<anonymous> (...FWUpdater\testReady.js:4:24)
>     at Module._compile (module.js:635:30)
>     at Object.Module._extensions..js (module.js:646:10)
>     at Module.load (module.js:554:32)
>     at tryModuleLoad (module.js:497:12)
>     at Function.Module._load (module.js:489:3)
>     at Function.Module.runMain (module.js:676:10)
>     at startup (bootstrap_node.js:187:16)
>     at bootstrap_node.js:608:3

reconbot, 2017-12-17:
> at Object. (E:\Projects\STM32\RemoteSwitch\FWUpdater\testReady.js:4:24)
> at Object. (...FWUpdater\testReady.js:4:24)

I think we already fixed this in master but haven't cut a new release yet.
Try using delimiter instead of data for now.

saewave, 2017-12-17:
Delimiter is not acceptable for me, but I can wait for release.
Thank you,

mm-ns, 2018-01-06:
any news on this?

reconbot, 2018-02-03:
the fix is in master, the last release got delayed due to travis issues, will have another one out in a day or two

reconbot, 2018-02-05:
fixed in the latest release v6.0.5

closed 2018-02-05 by reconbot