serialport/node-serialport #1426: Error:  Error Resource temporarily unavailable Cannot lock port

evpro, 2017-12-20:
- __SerialPort Version__: 6.0.4
- __NodeJS Version__: v4.8.2
- __Operating System__ and __Hardware Platform__: Linux raspberrypi 4.9.59-v7+ #1047

- Have you checked the right __version__ of the __api docs?__: yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?no
- Are you using __Electron__ and have you checked the _Electron Docs_?:no

### Summary of Problem
Hello, I need some help with serialport here. I am building RC CAR for my personal project I am doing testing and got an issue here. When I do "run" function it will write the data but the second time I run it will display this message "Error Resource temporarily unavailable Cannot lock port" and port.isOpen is always says false which is right, but if I use port.close connection everything works correctly, anyway to keep the port open and keep waiting for commands and send them?

### Steps and Code to Reproduce the Issue
`process.stdin.resume();
process.stdin.setEncoding('utf8');

process.stdin.on('data', function (text) {
  if(text.trim() === 'run'){
        console.log('running commmand');
        sendMessage();
  }
});

function sendMessage(){
        var SerialPort = require('serialport');
        var port = new SerialPort('/dev/ttyS0', {baudRate: 9600 });
        if(!port.isOpen){
                port.on('open', function () {
                        console.log('port opened');
                        port.write('running\n', function(err){
                                if(err){
                                        console.log(err);
                                }else{
                                        console.log('Message sent successfully');
                                }
                                console.log(port.isOpen);
                                // port.close(function () {
                                //      console.log('port Closed.');
                                // });
                        });
                });
        }
}`

fivdi, 2017-12-20:
It should be possible to resolve this issue by restructuring things a little. Currently the program creates a new SerialPort object every time sendMessage is called. If a single SerialPort object is created when the program starts and this instances is used for sending all messages then it should work correctly.

evpro, 2017-12-20:
@fivdi I already tried to make it global var outside of sendMessage function but port.on not even calling at all, so thats why I had ti put inside function, strange. I need to keep the port open and send commands based on buttons I press.

fivdi, 2017-12-20:
The port.on should also be outside sendMessage. Maybe something like this (untested):
```js
var SerialPort = require('serialport');
var port = new SerialPort('/dev/ttyS0', {baudRate: 9600 });

port.on('open', function () {
  console.log('port opened');

  process.stdin.resume();
  process.stdin.setEncoding('utf8');

  process.stdin.on('data', function (text) {
    if (text.trim() === 'run') {
      console.log('running commmand');
      sendMessage();
    }
  });
});

port.on('error', function (err) {
  console.error('Hmm..., error!');
  console.error(err);
  process.exit(1);
});

function sendMessage() {
  port.write('running\n', function(err) {
    if (err) {
      console.log(err);
    }else{
      console.log('Message sent successfully');
    }
  }
}
```

evpro, 2017-12-20:
@fivdi this is correct, that's what I done... thank you for your help. it was bothering me and I did same thing as what you did.

fivdi, 2017-12-20:
@evpro I don't think I fully understand. Is everything working now or is there still an issue?

fivdi, 2018-01-07:
@evpro I'm going to assume that everything is ok here and go ahead and close the issue. If there are still problems the issue can be reopened.

closed 2018-01-07 by fivdi