serialport/node-serialport #1427: cant receive data

xuxiaoyu9510, 2017-12-22:
- __SerialPort Version__:   v6.0.4
- __NodeJS Version__:   v7.8.0
- __Operating System__ and __Hardware Platform__:   win7  64bit

- Have you checked the right __version__ of the __api docs?__:[README.md](https://github.com/node-serialport/node-serialport/blob/v6.0.4/README.md#module_serialport--SerialPort+write)
- Are you using __Electron__ and have you checked the _Electron Docs_?:   no

```
import * as SerialPort from 'serialport'

let Ready = SerialPort.parsers.Ready;

let port = new SerialPort('COM4');

let parser = port.pipe(new Ready({delimiter: 'READY'}));

// const sendData = '0R0\n'
port.on('open', function() {
  port.write(Buffer.from('0R0\n') , function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
});

parser.on('data', function(data) {
  console.log('Data: ', data);
})

port.on('error', function(err) {
  console.log('Error: ', err.message);
})

```

I want to receive data when i send the instruction,but nothing.And i dont know why.

buuuudzik, 2018-01-08:
I have the same situation and I know that this is the problem of controlling RTS pin in UART. I've tried with change of such parameters like options.rts in serialport but nothing helped and still RTS pin isn't controlled apropriately.

// serialPort.set([options], [callback])
// [options.rts] | Boolean | true

reconbot, 2018-08-29:
I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2018-08-29 by reconbot

976032412, 2018-09-07:
@xuxiaoyu9510  解决了么  我也遇到类似问题（Have you solved it? I have similar problems.）