serialport/node-serialport #1428: Having issues communicating through RS232 interface

lemonbuzz, 2017-12-24:
- __SerialPort Version__: 6.0.4
- __NodeJS Version__: v8.6.0
- __Operating System__ and __Hardware Platform__: Ubuntu 16.04

- Have you checked the right __version__ of the __api docs?__: Yes
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? No
- Are you using __Electron__ and have you checked the _Electron Docs_?: No

### Summary of Problem
So i'm a total noob in regards to serial communication, so please bare with me if this is not worthy of opening an issue because it might just stem from my lack of understanding...
I am developing an EV controller and the commands I have to send to the electrical stuff has to ideally go through the RS232/COM port. During the development of the app, I have used the USB interface with an FTDI and I did not encounter any problems. I'm currently wrapping things up and decided to try out the actual physical interface that'll be used in production.

When i send the commands, i receive gibberish and strange characters....The only thing I have changed is the port name, from /dev/ttyUSB0 to /dev/ttyS0(which should be the correct port).
This is the hardware that will be used:
http://www.aaeon.com/en/p/embedded-computers-boxer-6403

Am I missing anything? Is there anything else other than the port name that should be changed when transitioning from USB/FTDI to COM Port/RS232 interface?

Thanks for the help! :)

### Steps and Code to Reproduce the Issue

reconbot, 2017-12-24:
That's really cool! I can't say for sure you're doing something wrong, your reasoning is sound. Bad looking data is usually a baudrate issue. Error reporting on Linux when setting a baudrate is poor, I would guess the actual hardware doesn't support the baudrate you're using. I'd dial back to 9600 test there and see what rates are supported on both your test and production devices.

lemonbuzz, 2018-01-03:
Thank you very much for the response!
I have already looked into the baudrate, I tried to dial it back to 9600 and increasing with each try, the result varies somewhat, but it's gibberish and i'm not getting the expected behaviour on the other end, regardless of the responses, so i'm guessing that even when it reads my messages, it gets gibberish too. For example, one command is suppose to toggle a relay, but it's just not happening when i'm using RS232.

To clarify, the test and production devices are the same on the controller side of things. It simply has TX/RX pins which I connect either the USB/FTDI pinouts or the RS232 data pins. Everything works fine with the USB/FTDI solution, but I have to get it working with the Rs232 interface and i'm just lost for solutions.

johnmanko, 2018-01-04:
Just a few thoughts. Can you verify the device permissions are the same for each?  Is it possible to run using `sudo` and do you see a difference?  Are you forcing any settings such as databits or parity?  What values are you using?

closed 2018-08-29 by reconbot