serialport/node-serialport #1429: Unable to do multiple writes

Cedric-Lefebvre, 2017-12-25:
Hello, i'm facing a problem when sending multiple sms at the same time, but for one, everything is ok.
When sending instead of twoo separate messages, i get only one message like:

AT+CMGS="XXXXXXXXX"
Test Message
Test Message

There is my code:

recipient.forEach(function(item) {
     send(port, item.tel,message);
})

function send(serial, toAddress, message) {
  
    setTimeout(function(){
        serial.write('AT+CMGF=1\r')
        setTimeout(function(){
            serial.write('AT+CMGS=\"'+ toAddress +'\"\r')
            setTimeout(function(){
                serial.write(message+'\r')
                setTimeout(function(){
                    serial.write('\x1A')
                }, 100);
            }, 100);
        }, 100);
    }, 100);

}

Everything is in last version, i tried to use the Drain functions etc, but nothing worked. Any help would be appreciated.

fivdi, 2017-12-28:
I would imagine that this is happening because the following loop from the program tries to send all text messages at the same time:
```js
recipient.forEach(function(item) {
  send(port, item.tel,message);
})
```
If the program is refactored to send the text messages one by one and to wait for each individual text message to be sent before the next text message is sent then it's more likely to work correctly.

Cedric-Lefebvre, 2018-01-04:
Thank you really, found the answer:

destinataire.delayedForEach(function(item){
                send(port, item.tel,result[0].message);
                list.push(item.tel);
            }, 1800);

Array.prototype.delayedForEach = function(callback, timeout, thisArg){
    var i = 0,
        l = this.length,
        self = this,
        caller = function(){
            callback.call(thisArg || self, self[i], i, self);
            (++i < l) && setTimeout(caller, timeout);
        };
    caller();
};

closed 2018-01-04 by Cedric-Lefebvre